goecharger/go-eCharger-API-v2 #124: trx setzen: falscher Wert wird übernommen (-1)

4homeassistant, 2023-03-19:
Wenn ich ```trx``` setzte wird immer ein um 1 niedrigerer Wert übernommen. Setze ich beispielsweise ```trx``` auf ```1``` wird ```0``` übernommen. Das Verhalten tritt sowohl über MQTT als auch über die lokale http-API auf.

```
http://xxx.xxx.xxx.xxx/api/set?trx=1
{"trx":true}

http://xxx.xxx.xxx.xxx/api/status?filter=trx
{"trx":0}
```

0xFEEDC0DE64, 2023-03-20:
0 wird für die app verwendet, 1-10 für die 10 rfid karten

4homeassistant, 2023-03-20:
Ja richtig Daniel. Lässt sich auf unseren beiden Chargern beliebig reproduzieren. Auf beiden läuft aktuell die 055.5 Beta. Das Verhalten besteht aber seit der 055.0.

0xFEEDC0DE64, 2023-03-28:
konnte den bug in der api key implementierung finden und beheben, beim nächsten firmware update ist das dabei!

closed 2023-03-28 by 0xFEEDC0DE64

4homeassistant, 2023-03-28:
Top! Danke Daniel 😊