goecharger/go-eCharger-API-v2 #125: MQTT TLS error / how to provide server certificate

denis-martin, 2023-03-19:
I'm trying to get MQTT via TLS working:

- Topic: mqtts://\<user>:\<pass>@\<host>:1883
- mqss ("MQTT skipServerVerification") set to true

As server, mosquitto is used and in the server logs, I can see connection attempts from the wallbox:
```
1679252871: New connection from 192.168.x.x:52874 on port 1883.
1679252871: OpenSSL Error[0]: error:14037419:SSL routines:ACCEPT_SR_KEY_EXCH:tlsv1 alert access denied
1679252871: OpenSSL Error[1]: error:140370E5:SSL routines:ACCEPT_SR_KEY_EXCH:ssl handshake failure
1679252871: Client <unknown> disconnected: Protocol error.
```

If I'm not mistaken, it looks like go-eCharger only supports TLSv1 which is rejected by the server, is that right?

In addition, I failed to set the MQTT server certificate via the API: 
```
$ curl -vvv 'http://192.168.178.105/api/set?msc="-----BEGIN%20CERTIFICATE-----%0...."'
 HTTP/1.1 414 URI Too Long
< Content-Type: text/html
< Content-Length: 15
< 
* Connection #0 to host 192.168.178.105 left intact
URI is too long
```
Can you provide an example on how to set a certificate? It seems like providing the certificate's content as query parameter does not work since it generates an URI that is too long.

0xFEEDC0DE64, 2023-03-28:
ok maybe our skipServerValidation is not working anymore, will check that

ratio91, 2023-04-05:
I have the exact same error.

Funzinator, 2023-05-04:
Any idea how to set the certificate via GET API (since it's too long) or how to properly disable certificate validation as a workaround?

poggenpower, 2023-12-30:
I have the same issue. It would be nice is there is a little bit more verbose Docs how to setup TLS based MQTT connections.

xieve, 2024-09-01:
Also having this issue with a self-signed certificate, having tried providing server certificate, CA certificate and both without any luck. Also having the same mosquitto error message as above, which to me sounds like the charger is trying to connect via TLSv1? I cannot get OCPP working with my certs either, but disabling server certificate validation does work there.