goecharger/go-eCharger-API-v2 #126: Go-e Charger rejects OCPP connection from EVCC

ChristophWatt, 2023-03-23:
Hello,

I have connected a Go-e Charger via OCPP to [EVCC](https://github.com/evcc-io/evcc) on Linux, but the Go-e charger rejects the connection every time.

How to reproduce:
- Connected a Go-e Charger via OCPP to EVCC
- Restart the Go-e charger to initialize the connection process
- The logs should show something similar to this:
`received JSON message from go-eCharger_205643: [3,"4103645269",{"status":"Rejected"}]`

I also attached the complete log file: [OCPP.log](https://github.com/goecharger/go-eCharger-API-v2/files/11049993/OCPP.log)

0xFEEDC0DE64, 2023-03-28:
it would help if we could add custom error messages to the rejected response, where I could write why I rejected something. But I am not sure if this would break the ocpp protocol.

What we can do now is you write a message to our support and tell them your serial number and ask for our OCPP remote debugging to be enabled, then I can see the reason for the reject message. Maybe we have to fix something in the ocpp implementation...

you can also send your serial number to my email adress

andig, 2023-03-28:
@0xFEEDC0DE64 if I can help from evcc side let me know. We can always run an Ocpp Server if you want to connect against.

0xFEEDC0DE64, 2023-03-28:
ok sure if you provide me with a ocpp url, i will connect

andig, 2023-03-29:
Could you contact info@evcc.io? Would need to do this over the weekend...

ChristophWatt, 2023-03-30:
Anything I can help with or is it better to not involve to many?

thomas-hutterer-tik, 2023-04-04:
Hi @0xFEEDC0DE64 please call me on 0699 18790722. I would love to get this resolved and I am at the location now where I can test the charger. Thanks Thomas

0xFEEDC0DE64, 2023-04-04:
sure, give me another hour or so, then I will call

andig, 2023-04-04:
You can also call me if necessary, @thomas-hutterer-tik has the details

0xFEEDC0DE64, 2023-04-04:
the firmware had a bug, that if the meter values buffer was empty it could not send the meter values. I changed the behaviour so that on request, we send a single meter values entry and do not change the interval of normally sent meter values (MeterValueSampleInterval and ClockAlignedDataInterval). We will test the new firmware with evcc soon

0xFEEDC0DE64, 2023-04-21:
Check out the new beta firmware release 055.6, it has been released!

slev1n, 2023-07-02:
Hi @0xFEEDC0DE64 ,
i also get "Rejected" if evcc want to change the state:
`Jul  2 22:27:40 raspberrypi evcc[7585]: [ocpp  ] INFO 2023/07/02 22:27:40 chargepoint connected, registering: goe123456
Jul  2 22:27:40 raspberrypi evcc[7585]: [ocpp  ] TRACE 2023/07/02 22:27:40 sent JSON message to goe123456: [2,"3526263417","GetConfiguration",{}]
Jul  2 22:27:40 raspberrypi evcc[7585]: [ocpp  ] TRACE 2023/07/02 22:27:40 received JSON message from goe123456: [3,"3526263417",{"configurationKey":[{"key":"AllowOfflineTxForUnknownId","readonly":false,"value":"false"},{"key":"GetConfigurationMaxKeys","readonly":true,"value":"99"},{"key":"ClockAlignedDataInterval","readonly":false,"value":"0"},{"key":"LocalAuthorizeOffline","readonly":false,"value":"true"},{"key":"HeartbeatInterval","readonly":false,"value":"60"},{"key":"LightIntensity","readonly":false,"value":"17"},{"key":"MeterValuesSampledData","readonly":true,"value":"Current.Import,Current.Offered,Energy.Active.Import.Register,Frequency,Power.Active.Import,Power.Factor,Power.Offered,Temperature,Voltage"},{"key":"MeterValuesSampledDataMaxLength","readonly":true,"value":"10"},{"key":"MeterValueSampleInterval","readonly":false,"value":"180"},{"key":"NumberOfConnectors","readonly":true,"value":"1"},{"key":"ConnectorPhaseRotation","readonly":true,"value":"0.RST,1.RST"},{"key":"ConnectorPhaseRotationMaxLength","readonly":true,"value":"3"},{"key":"SupportedFeatureProfiles","readonly":true,"value":"Core,FirmwareManagement,SmartCharging,RemoteTrigger"},{"key":"SupportedFeatureProfilesMaxLength","readonly":true,"value":"6"},{"key":"UnlockConnectorOnEVSideDisconnect","readonly":false,"value":"true"},{"key":"WebSocketPingInterval","readonly":false,"value":"10"},{"key":"LocalAuthListEnabled","readonly":false,"value":"true"},{"key":"LocalAuthListMaxLength","readonly":true,"value":"10"},{"key":"SendLocalListMaxLength","readonly":true,"value":"10"},{"key":"ChargeProfileMaxStackLevel","readonly":true,"value":"20"},{"key":"ChargingScheduleAllowedChargingRateUnit","readonly":true,"value":"Current"},{"key":"ChargingScheduleMaxPeriods","readonly":true,"value":"20"},{"key":"ConnectorSwitch3to1PhaseSupported","readonly":true,"value":"true"},{"key":"MaxChargingProfilesInstalled","readonly":true,"value":"20"},{"key":"AccessControl","readonly":true,"value":"Open"}]}]
Jul  2 22:27:40 raspberrypi evcc[7585]: [ocpp  ] TRACE 2023/07/02 22:27:40 AccessControl (r/o): Open
Jul  2 22:27:40 raspberrypi evcc[7585]: [ocpp  ] TRACE 2023/07/02 22:27:40 AllowOfflineTxForUnknownId (r/w): false
Jul  2 22:27:40 raspberrypi evcc[7585]: [ocpp  ] TRACE 2023/07/02 22:27:40 ChargeProfileMaxStackLevel (r/o): 20
Jul  2 22:27:40 raspberrypi evcc[7585]: [ocpp  ] TRACE 2023/07/02 22:27:40 ChargingScheduleAllowedChargingRateUnit (r/o): Current
Jul  2 22:27:40 raspberrypi evcc[7585]: [ocpp  ] TRACE 2023/07/02 22:27:40 ChargingScheduleMaxPeriods (r/o): 20
Jul  2 22:27:40 raspberrypi evcc[7585]: [ocpp  ] TRACE 2023/07/02 22:27:40 ClockAlignedDataInterval (r/w): 0
Jul  2 22:27:40 raspberrypi evcc[7585]: [ocpp  ] TRACE 2023/07/02 22:27:40 ConnectorPhaseRotation (r/o): 0.RST,1.RST
Jul  2 22:27:40 raspberrypi evcc[7585]: [ocpp  ] TRACE 2023/07/02 22:27:40 ConnectorPhaseRotationMaxLength (r/o): 3
Jul  2 22:27:40 raspberrypi evcc[7585]: [ocpp  ] TRACE 2023/07/02 22:27:40 ConnectorSwitch3to1PhaseSupported (r/o): true
Jul  2 22:27:40 raspberrypi evcc[7585]: [ocpp  ] TRACE 2023/07/02 22:27:40 GetConfigurationMaxKeys (r/o): 99
Jul  2 22:27:40 raspberrypi evcc[7585]: [ocpp  ] TRACE 2023/07/02 22:27:40 HeartbeatInterval (r/w): 60
Jul  2 22:27:40 raspberrypi evcc[7585]: [ocpp  ] TRACE 2023/07/02 22:27:40 LightIntensity (r/w): 17
Jul  2 22:27:40 raspberrypi evcc[7585]: [ocpp  ] TRACE 2023/07/02 22:27:40 LocalAuthListEnabled (r/w): true
Jul  2 22:27:40 raspberrypi evcc[7585]: [ocpp  ] TRACE 2023/07/02 22:27:40 LocalAuthListMaxLength (r/o): 10
Jul  2 22:27:40 raspberrypi evcc[7585]: [ocpp  ] TRACE 2023/07/02 22:27:40 LocalAuthorizeOffline (r/w): true
Jul  2 22:27:40 raspberrypi evcc[7585]: [ocpp  ] TRACE 2023/07/02 22:27:40 MaxChargingProfilesInstalled (r/o): 20
Jul  2 22:27:40 raspberrypi evcc[7585]: [ocpp  ] TRACE 2023/07/02 22:27:40 MeterValueSampleInterval (r/w): 180
Jul  2 22:27:40 raspberrypi evcc[7585]: [ocpp  ] TRACE 2023/07/02 22:27:40 MeterValuesSampledData (r/o): Current.Import,Current.Offered,Energy.Active.Import.Register,Frequency,Power.Active.Import,Power.Factor,Power.Offered,Temperature,Voltage
Jul  2 22:27:40 raspberrypi evcc[7585]: [ocpp  ] TRACE 2023/07/02 22:27:40 MeterValuesSampledDataMaxLength (r/o): 10
Jul  2 22:27:40 raspberrypi evcc[7585]: [ocpp  ] TRACE 2023/07/02 22:27:40 NumberOfConnectors (r/o): 1
Jul  2 22:27:40 raspberrypi evcc[7585]: [ocpp  ] TRACE 2023/07/02 22:27:40 SendLocalListMaxLength (r/o): 10
Jul  2 22:27:40 raspberrypi evcc[7585]: [ocpp  ] TRACE 2023/07/02 22:27:40 SupportedFeatureProfiles (r/o): Core,FirmwareManagement,SmartCharging,RemoteTrigger
Jul  2 22:27:40 raspberrypi evcc[7585]: [ocpp  ] TRACE 2023/07/02 22:27:40 SupportedFeatureProfilesMaxLength (r/o): 6
Jul  2 22:27:40 raspberrypi evcc[7585]: [ocpp  ] TRACE 2023/07/02 22:27:40 UnlockConnectorOnEVSideDisconnect (r/w): true
Jul  2 22:27:40 raspberrypi evcc[7585]: [ocpp  ] TRACE 2023/07/02 22:27:40 WebSocketPingInterval (r/w): 10
Jul  2 22:27:40 raspberrypi evcc[7585]: [ocpp  ] TRACE 2023/07/02 22:27:40 sent JSON message to goe123456: [2,"1612164215","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[... cut ...]
Jul  2 22:28:00 raspberrypi evcc[7585]: [ocpp  ] TRACE 2023/07/02 22:28:00 sent JSON message to goe123456: [2,"2558598256","RemoteStartTransaction",{"connectorId":1,"idTag":"evcc","chargingProfile":{"chargingProfileId":1,"stackLevel":0,"chargingProfilePurpose":"TxProfile","chargingProfileKind":"Relative","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":16}]}}}]
Jul  2 22:28:00 raspberrypi evcc[7585]: [ocpp  ] TRACE 2023/07/02 22:28:00 received JSON message from goe123456: [3,"2558598256",{"status":"Rejected"}]
`

No matter if dummy card is activate with id 'evcc' or deactivated. Generell authentification is disabled.

andig, 2023-07-02:
Formatting the logs would make them readable ;)

slev1n, 2023-07-02:
@andig thx for the hint. I uploaded the log file

0xFEEDC0DE64, 2023-07-03:
at the end of the log I see a Rejected response to a RemoteStartTransaction message,

what is the status of the charger in that moment? has another transaction already been started?

slev1n, 2023-07-03:
No other transaction, clean reboot of the charger and evcc.

slev1n, 2023-07-04:
I updated and restartet charger and pi and now it works:). 
Just run into another open issue [OCPP: Current limit ignored when suspended charging session is resumed](https://github.com/goecharger/go-eCharger-API-v2/issues/155)

closed 2024-09-26 by erxbout