goecharger/go-eCharger-API-v2 #131: Low-Power Standby / "Power-On" Schedule / Turn-Off over night

franky436, 2023-04-05:
Der go-e braucht an die 10W. Dabei betreibe ich nur PV Überschussladung, wenn tagsüber die Sonne scheint.

Habe nun überlegt, auf L1 eine Zeitschaltuhr vorzuschalten, oder ein Shelly der weniger als 1W braucht. - Eleganter wäre natürlich, wenn man eine Art "Zeitplan" für "Power-On" definieren könnte, und der go-e sich über Nacht ausschaltet, oder eine Art "lo-power" Modus.

Will nicht geizig klingen, bei den aktuellen Strompreisen läppert sich das "Standby" aber schon nennenswert: 
20h×0,010W×365dx0,55€ = 40,15/a = 401,50€ in 10 Jahren, nur für "Standby".

p.s. Möchte unbedingt auch ein DICKES Lob loswerden! Die Software & API vom go-e sind eine der besten, mit denen ich in den letzten 3 Jahren zu tun hatte. - Habe mich in den letzten 36 Monaten mit diversen Geräten über APIs, Modbus, REST, Websockets, et cetera beschäftigt.... und ganz ehrlich: go-e ist richtig, richtig cool und empfehle ich jedem weiter. Danke dafür, dass Software auch die nötige Liebe bekommt, und nicht nur Hardware!

peterpoetzi, 2023-04-11:
Hi, es sollten deutlich unter 10W sein, wir haben bei ausgeschalteten LEDs 1,9W gemessen und diesen Wert auch im Datenblatt stehen. Mit allen LEDs an sind es maximal 4,2W.

Ich denke bei 1,9W, also etwa 17kWh im Jahr, zahlt es sich nicht wirklich aus einzelne Verbraucher abzuschalten, auch wenn es technisch natürlich möglich wäre WLAN und RFID abzuschalten, da man hier nur mehr ein paar Milliwatt sparen kann.

Vielen dank für das Lob :)

franky436, 2023-04-15:
Vielen Dank, Peter!  

Das mit den 10W habe ich ehrlicherweise nicht selbst nachgemessen, sondern online aufgeschnappt.

Die von Dir genannten Werte klingen wirklich in Ordnung.