goecharger/go-eCharger-API-v2 #132: The size of the tma response array changed inadvertedly

TeddeBaer, 2023-04-10:
The API v 2 documentation does not explicitly state the size of the array of the tma response ("temperature sensors"). In my case, the array size is usually 2, however every now and then, the response only contains one value. I haven't found a way to reproduce it systematically, but when the cable is connected to the EV, this happens usually during a timeframe from 5 minutes to 2 hours.

This is a trace from my OpenHAB log:
> 2023-04-10 02:05:01.529 [TRACE] [internal.handler.GoEChargerV2Handler] - GET Response: {"alw":true,"dwo":null,"eto":527952,"amp":16,"frc":2,"car":2,"err":0,"cbl":20,"wh":16481.01,"trx":null,"fwv":"055.0","tma":[25,22.5],"nrg":[227.54,225.99,226.61,2.79,15.314,15.39,15.466,3484.561,3471.439,3503.163,0,10459.16,100,99.81176,99.95472,100],"psm":2}
> 2023-04-10 02:05:06.650 [TRACE] [internal.handler.GoEChargerV2Handler] - GET Response: {"alw":true,"dwo":null,"eto":527967,"amp":16,"frc":2,"car":2,"err":0,"cbl":20,"wh":16495.62,"trx":null,"fwv":"055.0","tma":[25],"nrg":[227.85,226.61,227.23,2.79,15.276,15.352,15.428,3485.241,3472.774,3502.421,0,10460.44,100,99.82345,99.90636,100],"psm":2}

0xFEEDC0DE64, 2023-04-11:
ok that's interesting, this should always be size 2.

Are you running an ancient old firmware version or is it up to date?

0xFEEDC0DE64, 2023-04-11:
no charger reported a tma array with different size to the cloud in the last 7 days, maybe just the http api is affected or you are modifying the value afterwards? for example null values in the array would mean a readout error of the temperature sensor on the i2c bus

TeddeBaer, 2023-04-11:
My firmware version is 055.0, which is the newest non beta version available in my go-eCharger app.

To my understanding, this is the response from the GET request. The value is thus unaltered. I attach the full trace log below:
```
2023-04-10 02:04:56.296 [TRACE] [internal.handler.GoEChargerV2Handler] - GET URL = http://192.168.1.130/api/status?filter=mod,psm,trx,alw,tma,wh,dwo,frc,nrg,car,amp,err,cbl,eto,fwv
2023-04-10 02:04:56.411 [TRACE] [internal.handler.GoEChargerV2Handler] - GET Response: {"alw":true,"dwo":null,"eto":527937,"amp":16,"frc":2,"car":2,"err":0,"cbl":20,"wh":16466.37,"trx":null,"fwv":"055.0","tma":[25,22.5],"nrg":[227.23,225.99,227.23,2.79,15.314,15.39,15.428,3481.722,3471.754,3501.517,0,10454.99,100,99.82082,99.88056,100],"psm":2}
2023-04-10 02:05:01.438 [TRACE] [internal.handler.GoEChargerV2Handler] - GET URL = http://192.168.1.130/api/status?filter=mod,psm,trx,alw,tma,wh,dwo,frc,nrg,car,amp,err,cbl,eto,fwv
2023-04-10 02:05:01.529 [TRACE] [internal.handler.GoEChargerV2Handler] - GET Response: {"alw":true,"dwo":null,"eto":527952,"amp":16,"frc":2,"car":2,"err":0,"cbl":20,"wh":16481.01,"trx":null,"fwv":"055.0","tma":[25,22.5],"nrg":[227.54,225.99,226.61,2.79,15.314,15.39,15.466,3484.561,3471.439,3503.163,0,10459.16,100,99.81176,99.95472,100],"psm":2}
```
The source code achieving this is: (Snip from the GoECharger binding in OpenHAB)
```
        String urlStr = getWriteUrl(key, value);
        logger.trace("POST URL = {}", urlStr);

        try {
            HttpMethod httpMethod = HttpMethod.GET;
            ContentResponse contentResponse = httpClient.newRequest(urlStr).method(httpMethod)
                    .timeout(5, TimeUnit.SECONDS).send();
            String response = contentResponse.getContentAsString();

            logger.trace("{} Response: {}", httpMethod.toString(), response);
```
Discussion about this issue can be found [here (OpenHAB community)](https://community.openhab.org/t/go-echarger-binding-stops-working-after-a-few-days-perhaps-issue-with-firmware-version-0-55/145070) and [here (Binding pull request in GitHub)](https://github.com/openhab/openhab-addons/pull/14632).