goecharger/go-eCharger-API-v2 #133: Export fup via MQTT

FransOv, 2023-04-12:
fup (usePvSurplus) is not exported via MQTT so cannot be monitored and set via e.g. Home Assistant. Could you export fup in the same way as for instance acs is exported: set at startup with retian true and whenever fup is updated with retain false?

0xFEEDC0DE64, 2023-04-13:
Ok will do, thanks for letting us know

0xFEEDC0DE64, 2023-04-13:
I think the reason why we reduced the number of api keys from the mqtt connection was that we are sending all values in one go, and having 300 pending mqtt requests overflows our memory (and creashes the charger). Also customers complained about too many topics being published.
The memory problem will be fixed by a buffering system, which waits for the first 10 mqtt commands being completed before putting more one the queue,
but about having too many api keys we don't have an idea yet how to fix this. We might implement a custom interval at which we broadcast data on mqtt and also maybe an api key filter like the http api already has got. Lets see

FransOv, 2023-04-13:
It would also help if you blocked the mqtt messages on frequently changing values that are not really ver useful for the monitoring and control like local and utc time and milliseconds since last boot. Also you could limit the amount of mqtt messages for some values by providing them in less precision and only publishing them on a significant change. Net frequency  and voltage levels in three decimals is a bit of an overkill.

povlhp, 2023-04-18:
I completely agree. I have multiple sensors with way too frequent updates.
If we could set MQTT update frequency for frequently changing values it would be great.  Somewhere in the 2-60 seconds range or whatever lower limit makes sense to implement. Configurable would be great.   My shelly EM3 spams my server as well - that is bad. And my own Solutronic script runs every 5 seconds reporting solar production - but only 2 values (with attributes) - I do it that often as clouds changes solar power quite a lot, and for short times. 

Currently, my solar charging logic runs every 2 minutes, but I would likely do 10s updates on the discrete values.

Some values should be pushed right away - but that is the state change values. Car connected, charging started/stopped. FRC for forced 1-phase etc. These are supposed to change rarely.

0xFEEDC0DE64, 2023-04-21:
fup (and some other) are now available as MQTT in the firmware release 055.6

FransOv, 2023-04-21:
I upgraded to version 0.55.6 Beta. There are a lot more parameters when I request the status via the http interface. Hope you don't overtax the esp32. It's a wonderful little SOC, I've got tens of them around the house doing all sorts of useful (and silly) things, but they are still limited.

0xFEEDC0DE64, 2023-04-26:
Then please start using the `/api/status?filter=alw,acu,rssi` filter to only get the parameters you need, will save you time and avoids possible crashes :+1: 

I have spent the last 3 years of my life getting the 400 total api keys to run on our products and I think the ESP32 such a beast, but I will keep checking the stability of the new firmwares.

povlhp, 2023-04-27:
Did you update the MQTT publish as well ? MQTT explorer shows only few regular updates. 
Or does it suppress non-changed values ?
Another idea, if you can have a filter in API call, then maybe MQTT filter as well - if it is not you who limits number of pushed messages. 

Things works great apart from too aggressive 3-phase on startup after pause in eco PV mode

0xFEEDC0DE64, 2023-04-27:
we are only pushing new api keys when the value changes, and we flag them with retain. This behaviour was never changed in the last 12 months. Are you expecting a certain api keys value changes but it does not on MQTT?

povlhp, 2023-04-28:
All working great in this respect. Just there was some discussion about MQTT overload. 
Maybe somebody wanted to be able to turn off time stamps. They have little value on most logic we implement.