goecharger/go-eCharger-API-v2 #134: OCPP TriggerMessage rejects "MeterValues"

tflanitzer, 2023-04-12:
Sending an OCPP TriggerMessage with RequestedMessage "MeterValues" always leads to a "Rejected" response and no meter values are sent. This is regardless of which connectorId is specified. Triggering works for other requests, at least for "BootNotification" and "StatusNotification". 

Is this behaviour intended or is it a bug? The documentation simply lists TriggerMessage as "implemented".

0xFEEDC0DE64, 2023-04-13:
This has been fixed in a development version of the firmware, which has not been tested yet fully. I can give you the beta firmware if you provide the serial number. If you downgrade with the app again, you cannot go back to the beta firmware on your own, I would have to do it again.

tflanitzer, 2023-04-13:
Do you have any rough estimate on when the new firmware will be released? Thanks for the offer with the beta version but it is enough for me to know that it will work soon.

0xFEEDC0DE64, 2023-04-20:
today or tomorrow

0xFEEDC0DE64, 2023-04-21:
Release is ready in the app as beta: 055.6