goecharger/go-eCharger-API-v2 #136: 'alw' in v2 read only?

jumlauft, 2023-04-15:
In v1 habe ich die Ladung via 'alw=1' aktiviert. Allerdings lässt sich der in v2 nicht mehr setzen. 
urllib.request.urlopen("https://SERIAL.api.v3.go-e.io/api/set?token=TOKEN&alw=1")
Mach ich etwas falsch oder wurde das geändert? Falls geändert, wie kann in v2 die Ladung aktiviert werden?

0xFEEDC0DE64, 2023-04-15:
Aus welchem Grund wird denn aktuell nicht geladen?

Fehlt die Authorisierung/Transaction, dann muss die gestartet werden: trx=0

Soll sämtliche smarte Ladelogik (wie zb Awattar, Überschuss, Next-Trip, usw.) überschrieben werden und einfach mit eingestellten Strom-Limit bis zum Abstecken des Autos geladen werden, kann man mit dem Force-State das Laden erzwingen: frc=2 (0=Neutral, 1=ForceOff, 2=ForceOn)

alw ist nur das Ergebnis der komplexen Ladelogik, ob nun geladen werden darf oder nicht. Bei V1 haben wir blöderweise das Setzen des api keys mit angeboten und das ließ sich mit der steigenden Komplexität der Ladelogik nicht mehr implementieren. Die V1 legacy-api im V3/V4 charger setzt den Force State auf ForceOn oder ForceOff, wenn alw gesetzt wird, was dann oft zu Verwunderungen geführt hat, wenn zb maximaler Awattar Tarifpreis eingestellt war und er trotzdem geladen hat.

jumlauft, 2023-04-16:
Top, damit funktioniert es. Vielen Dank für die schnelle Antwort!

0xFEEDC0DE64, 2023-04-21:
dann close ich das mit der Erfolgsmeldung

closed 2023-04-21 by 0xFEEDC0DE64