goecharger/go-eCharger-API-v2 #139: WLAN verbindet sich mit dem schwächsten Signal und will nicht los lassen

gyroslav, 2023-04-19:
Hallo zusammen,
ich hab 4 Fritzbox-Repeater im Mash im Haus, einer davon hängt in der Garage, Luftlinie 4 Meter von der Go-E entfernt. 
Alle paar Tage hängt sich die Go-E an das schwächste Signal von einem der anderen Repeater und will nicht mehr loslassen. 
Abhilfe hilft ein reboot oder manueller wechsel in der APP. Dabei muss ich aber das Passwort jedes Mal neu eingeben, alswenn das immer ein neues Netz für die Go-E wäre. 
In der APP sehe ich alle 4 Signale mit dem gleichen Namen, und einer davon hat ein GUTES Signal. Die anderen ein "!". 

Weiß jemand einen Rat, außer den AP in der Garage aus dem Mash zu nehmen und eine eigene ID zu vergeben?

gyroslav, 2023-04-21:
Gibt es einen API-Key, der die Signalqualität repräsentiert? 
Dann kann ich das versuchen zu loggen...

0xFEEDC0DE64, 2023-04-21:
ja, ganz kreativer name des api keys: rssi

gyroslav, 2023-04-25:
heute wieder passiert, hab es nun im LOG drin. Werde eine Mail an support@go-e.com schreiben.

sqw534, 2025-04-15:
Hast Du vom Support dazu mal eine Antwort bekommen? Ich habe eine ähnliche Situation und leider ist der zweite AP gerade noch so erreichbar, aber nicht wirklich nutzbar und trotzdem verbindet sich der Go-E diesen immer mal wieder zu diesem.

gyroslav, 2025-04-16:
Es hat seit dem Problem diverse Firmware-Updates gegeben, mit denen es bei mir etwas besser wurde. 
Eine Zeitlang habe ich mir mit einem Script geholfen (Iobroker Blockly), das den RSSI Wert auswertet und wenn der zu schlecht wird, setzt das Script einen Reboot-Befehl ab. Natürlich nur, wenn gerade kein Ladevorgang aktiv ist. 

Mittlerweile habe ich aber mein Netzwerk in der Garage auch umgebaut. Hab eine gebrauchte ebay-Fritzbox per LAN-Kabel relativ mittig in der Garage hängen. Die deckt alle Shellys und die GO-E gut ab, das Signal ist stabiler als beim Repeater. 
Mit der Fritzbox ist der Wechsel auf einen AC im Haus noch nicht wieder aufgekommen.