goecharger/go-eCharger-API-v2 #141: Pricing in the App (Awatar mode)

0xFEEDC0DE64, 2023-04-20:
### Discussed in https://github.com/goecharger/go-eCharger-API-v2/discussions/140

<div type='discussions-op-text'>

<sup>Originally posted by **povlhp** April 19, 2023</sup>
I am located in Denmark, as for pricing mode, I can pick a plan among many electricity providers. This is basically not interesting.

In Denmark, we have network providers which are separate from electricity providers. And people usually have either fixed price, or variable price (nordpool spot + 0 - 2 cents). And of course we have 2 price zones (East/West) - but they are reflected today.

The network providers has different "transport" fees, so to make the data more usable, I would like to have the price reflect the Spot price + transport fees.

Transport in my area is:
DKK 0,1530  from 0-6
DKK 0,2296 from 6-17 and 21-00
DKK 0,5969 from 17-21

Other areas have 3+ DKK in transport fees 17-21.

Electricity price is anything between a slight negative number and typicall max 2 DKK (rare hours more).

Thus the transport fee makes a HUGE difference when average electricity price is around DKK 0.70.

Trabnsport fee is there to help people move consumption to the night or day.</div>

kalaws, 2023-04-26:
An adjacent issue is that PV mode, or the charge planner, could hopefully factor in the difference in export and import price that owes to certain fixed additional costs and VAT added to spot rate at import but not at export. For example, in Sweden, there is in practice a 0,06EUR subsidy per exported kWh. This means that, even though spot rates do not indicate this, sometimes it is more beneficial to export PV surplus and postpone charging instead to the evening, using imported grid power when there is no PV generation.

In the following example, even though there is a PV surplus and the spot rate is lower at noon, it is in fact best not to charge but postpone charging for later (instead exporting the surplus).
```
Hour    PV surplus   Spot rate   Export sub.  Import fee  | Fetched price     OPTIMAL GO-E BEHAVIOR
----------------------------------------------------------|-------------------------------------
12-13   YES          0,015       0,06        (0,04)       | 0,075 EUR (sell)  DO NOT CHARGE
20-21   NO           0,020+VAT  (0,06)        0,04        | 0,065 EUR (buy)   CHARGE

VAT=0,25%
```

povlhp, 2023-06-23:
kalaws,then we really should have the option to do some templating - Or at least have multiple fields.
Say use Awatar (Spot price) +/- fixed amount and percentage on import and different on export
But here in Denmark we have 3 time zones with different fixed fee. 00-06  06-17 17-21 and 21-00 we use the 06-17 timezone again. For a fixed price transport cost.

And when it comes down to everything, maybe it is better to use Home Assistant to control. There are some HA integrations with relevant calculations.

I think the box should do it for non-PV users.

When I sell electricity, I pay approx 40% tax and 0.008 DKK fee. But on some I will save electricity tax later.
If I use it myself, I pay a fixed price (tax of 0.40 DKK) approx 0.16 DKK
When I buy it, I pay full spot + 25% VAT + DKK 0.87 electricity tax + transport (0.15DKK to 0.50DKK - Some places up to 2+ DKK from 17-21) 

But, for each kWh I sell, I can avoid the  DKK 0.87/kWh on one kWh import. And if I export more than I buy over the year, then my surplus export gets a price adjustment up/down such that the few kWh will have an average price of DKK 0.40.

kalaws, 2023-06-28:
@povlhp did you get any further with this?

povlhp, 2023-06-28:
No. I Think a good solution would be to have either a custom URL or MQTT key that we could overwrite. Just need the specs of data as well. With Nordpol spot we have prices we can push for the next 24-48 hours.

kalaws, 2023-07-26:
> No. I Think a good solution would be to have either a custom URL or MQTT key that we could overwrite. Just need the specs of data as well. With Nordpol spot we have prices we can push for the next 24-48 hours.

Completely agree with this. Would be fantastic simply to push one's own tariff, so that go-e does all the rest of the planning. Much better than trying to manage the charging process itself externally. @peterpoetzi could this be achieved?

kalaws, 2023-08-21:
No updates on this?