goecharger/go-eCharger-API-v2 #143: Go-e charger gemini negative charge power in evccfor some secounds

djrepose, 2023-04-26:
When i start the charging over cvcc with the v2 api it appears allways 2 messages with negative charging power in mA. 
is it possible to fix that?
[image]

Emacee, 2023-08-12:
I have this same issue when using a Type 2 to Schuko adapter and connect a small load. The Go-E app and API show a negative value when using my bike charger on this adapter. Only when the load is above +/- 100Wh it shows the correct value. This also corrupts the OCPP session since it doesn't accept negative sessions. I contacted Go-E support but they didn't acknowledge this as a problem (yet) by arguing that the charging itself works. Hopefully someone here is able to fix this in the software of the charger.

bawe, 2024-03-19:
Same here with:
go-e Charger Gemini (Hardware V4/ Firmware 55.8)
Renault Zoe

mkalinna, 2024-03-25:
Same here with:
go-e Charger Gemini (Hardware V4/ Firmware 55.8)
Renault Twingo