goecharger/go-eCharger-API-v2 #145: Charger verliert beim Phasenwechsel Authentifizierungs Information

ddrrhh, 2023-04-29:
Bei meiner PV Laderegelung verliert der charger immer mal wieder die Authentifizierungsinformation.
Der Phasenwechsel wird durchgeführt mit:

frc:1  # Stoppe Ladung;
psm:2 bzw 1  # Stelle der Phasen ein
frc:0  # Starte Ladung.

Die Firmware des Laders ist 55.0. Das Problem ist aber auch schon mit den 54.x Firmwareversionen aufgetreten. Bei noch früheren Versionen habe ich es nicht beobachtet.
Die Autos sind  Zoe BJ 2019 bzw, 2020
Das Problem tritt nicht bei jedem Phasenwechsel auf. Ob es von der Richtung von 3 auf 1, oder von 1 auf 3 abhängt konnte ich noch nicht  ermitteln. 
Sollte der Phasenwechsel nur durch die Umschaltung von psm durchgeführt werden?

chk2902, 2023-06-29:
Gibt's hier irgend ein Update?

Ist das Problem ein Problem für folgende Situation (beim PV-Überschussladen)

- Benutzer hängt Auto an
- authentifiziert sich per RFID
- Ladung läuft, schaltet Phasen um, schaltet Ladeprozess ein/aus, ... (alles außer Auto abhängen)

0xFEEDC0DE64, 2023-06-30:
Habe eben einen Bug bei den V4 aufgedeckt, könnt ihr mir bestätigen, dass es nur auf V4 aufgetreten ist?

Nächste Beta Firmware sollte es für die V4 verbessern.

ddrrhh, 2023-06-30:
Ja richtig! Allerdings passiert es nicht bei jedem Phasenwechsel

ddrrhh, 2023-06-30:
Ja es ist eine V4

chk2902, 2023-07-01:
Danke - dann muss ich vorsichtshalber die Authentifizierung bis zum Abstecken merken.

ddrrhh, 2023-10-07:
Ich nutze seit einiger Zeit die neuste stabile Firmwareversion 55.7 und muss feststellen, das Verhalten ist eher schlimmer geworden. Die Authentifizierungsinformation geht, so weit ich sehen konnte bei jedem Phasenwechsel verloren, darüberhinaus verliert der Charger auch bei jedem Phasenwechsel auch die bisher geladene Energie. Und fängt nach einem Phasenwechsel wieder bei 0 an.

fre4242, 2023-10-13:
Ich kann das Problem mit 55.7 bestätigen. Wobei die Authentifizierung manchmal die Umschaltung überlebt. Meistens aber nicht.

cy-fre, 2024-01-19:
Das Problem ist bei mir mit 55.8 nicht mehr aufgetreten. Der issue kann meines Erachtens geschlossen werden.

peterpoetzi, 2024-01-19:
Wurde mit Firmware 55.8 für go-e Charger V3 und V4 (Gemini) gefixt

closed 2024-01-19 by peterpoetzi

hajohue, 2024-03-23:
Ich habe das gleiche Problem bei der HomeFix V3. Das Upgrade auf Firmware 55.8 brachte keine Änderung.
Mache ich etwas falsch oder heisst das, dass das Verhalten bei der HomeFix nicht gefixt ist / wird?
Danke für jede Hilfe.

0xFEEDC0DE64, 2024-04-02:
@hajohue wir bräuchten die seriennummer und einen genauen zeitraum, wo das passiert ist, gerne auch über den support. wir wollen das problem gefixt bekommen bei allen generationen des produkts

hajohue, 2024-04-02:
@0xFEEDC0DE64,
interessanter Weise habe ich einen Tag nach meinem Eintrag hier in der App zur Wallbox ein neues Update angezeigt bekommen. Angegeben war die Version 55.8, obwohl ich diese ja schon bereits installiert hatte. Ich habe dann das Update geladen und seither ist der Fehler nicht wieder aufgetreten. Ich beobachte das weiter.
Wenn Sie meine Seriennummer trotzdem benötigen, bitte ich um eine kurze Meldung. Gerne sende ich die dann über den Support.

hajohue, 2024-04-16:
Seit dem erneuten Update (55.8) vor zwei Wochen ist das Problem nicht wieder aufgetreten.