goecharger/go-eCharger-API-v2 #154: OCPP: "numberPhases" ignored in SetChargingProfile

tflanitzer, 2023-06-02:
While changing the amperage via the OCPP request SetChargingProfile works as expected, the included "numberPhases" is ignored. A switch from 3 to 1 or 1 to 3 never happens, the charger sticks with its initial number of phases. However, when done via the app, it does so immediately.

Here is an example request body:

`{
  "connectorId": 1,
  "csChargingProfiles": {
    "chargingProfileId": 1,
    "stackLevel": 1,
    "chargingProfilePurpose": "TxProfile",
    "chargingProfileKind": "Absolute",
    "chargingSchedule": {
      "startSchedule": "2023-06-02T00:00:00.3828555Z",
      "chargingRateUnit": "A",
      "chargingSchedulePeriod": [
        {
          "startPeriod": 0,
          "limit": 6.0,
          "numberPhases": 1
        }
      ]
    }
  }
}
`

I have tried it with TxProfile, TxDefaultProfile and even ChargePointMaxProfile. In the app I tried with the number of phases set to all possible values ("Automatisch", "1-Phase", "3-Phase"). The corresponding configuration value "ConnectorSwitch3to1PhaseSupported" reports as true.

Device: go-e Gemini 11kW (hardware version V4)
Firmware version: 055.6 Beta

0xFEEDC0DE64, 2023-06-06:
Thanks for reporting!
I have implemented numberPhases now and after some beta-testing in-house we will release the beta firmware. Timeframe: 1-2 weeks

0xFEEDC0DE64, 2023-06-13:
beta firmware 055.7 is available through the app