goecharger/go-eCharger-API-v2 #155: OCPP: Current limit ignored when resuming charging

tflanitzer, 2023-06-02:
When a charging session is suspended by setting a charging profile with current limit of 0 and is afterwards resumed by setting a current limit > 0, the latter limit is ignored and the charger resumes charging without limit. Subsequent charging profiles again work as expected.

The scenario in detail (the charger is set to 16A in the app):

1. go-e is charging
2. Charging is suspended with OCPP SetChargingProfile
```
{
  "connectorId": 1,
  "csChargingProfiles": {
    "chargingProfileId": 1,
    "stackLevel": 1,
    "chargingProfilePurpose": "TxProfile",
    "chargingProfileKind": "Absolute",
    "chargingSchedule": {
      "startSchedule": "2023-06-02T00:00:00.3828555Z",
      "chargingRateUnit": "A",
      "chargingSchedulePeriod": [
        {
          "startPeriod": 0,
          "limit": 0,
          "numberPhases": 1
        }
      ]
    }
  }
}

```
3. go-e stops charging and reports status "SuspendedEvse"
4. Charging is resumed with with OCPP SetChargingProfile
```
{
  "connectorId": 1,
  "csChargingProfiles": {
    "chargingProfileId": 1,
    "stackLevel": 1,
    "chargingProfilePurpose": "TxProfile",
    "chargingProfileKind": "Absolute",
    "chargingSchedule": {
      "startSchedule": "2023-06-02T00:00:00.3828555Z",
      "chargingRateUnit": "A",
      "chargingSchedulePeriod": [
        {
          "startPeriod": 0,
          "limit": 6,
          "numberPhases": 1
        }
      ]
    }
  }
}
```
5. go-e resumes charging with 16A instead of the requested 6A.

Device: go-e Gemini 11kW (hardware version V4)
Firmware version: 055.6 Beta

0xFEEDC0DE64, 2023-06-22:
I could not reproduce this issue with the newest firmware and steve backend, can we arrange a live call and testing session? I would enable remote-ocpp debugging functionalities on your charger to see all state transitions of the ocpp stack.

0xFEEDC0DE64, 2023-06-23:
Found another related bug, please test again with the next coming beta release (next days)

tflanitzer, 2023-06-23:
I will try to reproduce this issue next week and report back. If it still exists, we can certainly arrange a live call.

tflanitzer, 2023-07-25:
Could not reproduce either, seems to be fixed. Thank you.

closed 2024-09-26 by erxbout