goecharger/go-eCharger-API-v2 #156: car zeigt status 4 obwohl auto nicht voll geladen nach einem frc=1 (Überschussladen)

lyncker, 2023-06-04:
Ich setze das Laden aus mit frc=1, wenn ich nicht genug Strom vom Dach habe und mein Füllstand der Batterie zu klein wird. Der Status spring dann auf 4. Wenn ich das LAden des Autos am morgen fortsetzen möchte bleibt aber der status in 4 und es findet kein Laden statt. Erst wenn ich den Stecker vom Auto abziehe und wiede dran stecke gehts weiter und car wechselt in 2 . Woran liegt das?

herbert131, 2023-06-04:
Wie lange gibst du der WB Zeit? Typischerweise dauert es eine Weile bis die WB nach Start (frc=2) dann den Status auf car = 2 setzt

lyncker, 2023-06-04:
ja also gestenr abend so gegen 23 Uhr muss irgendwann der SOC der Batterie unter 40% gegangen sein, darauf hin wurde dann FRC auf 1 gesetzt und um 23:45 ging der status dann auf 4. Heut morgen wurde dann FRC auf 0 gestzt aber der Status verblieb in 4 und das laden setzte sich nicht fort. Ich habe dann manuell auch mal frc=2 gesetzt aber ein paar Minuten sah man keinen Ladestrom. Wie lange dauert es denn?

herbert131, 2023-06-04:
so ca. 10-15s, frc 0 bedeutet wohl, dass die WB automatisch das tut was eingestellt ist ( ...) mit frc 2 erzwingst du den Start. hast du den access control aktiviert (acs) ?

lyncker, 2023-06-04:
hmm ne ACS steht auf 0 ... ich finde dazu auch kaum was... wofür genau soll der sein?

herbert131, 2023-06-04:
ja die Dokumentation ist sehr sehr verbesserungswürdig....
mit ACS kannst z.B. per RFI das Laden freigeben, aber 0 ist OK
kommt natürlich auch auf den Auto an... mit meinem Peugeot gehts so wie beschrieben. 
Kannst du mit der Go-e app das Laden denn starten ? was steht denn als Status auf de APP ? evtl kannst dir daraus ableiten an was es hängt

lyncker, 2023-06-04:
ich hatte die app zwar offen , aber es sag so aus als wäre alles ok ... nur das Laden startete halt nicht... ich habe leider keinen Screenshot und muss mir das beim nächsten mal nochmal ansehen...

dakonga, 2023-06-12:
Das ist doch das gleiche Problem was ich unter #144 aufgemacht hatte. Bisher keine Reaktion.

phk0812, 2023-06-27:
Ich habe das selbe Problem, wenn ich mangels ausreichend PV-Strom das Laden unterbreche.
Allerdings kann ich das Laden problemlos wieder starten.
Möglicherweise da ich die Aktivierung per RFID nutze.

Idealerweise gäbe es die Möglichkeit eine Ladepause zu nutzen.
Die kann ja, falls das Autos dies nicht erlaubt, eine solche Pause simulieren.
Und entsprechende Werte bei modelStatus setzen.

gyroslav, 2023-07-07:
Manche Autos gehen in Tiefschlaf, wenn sich längere Zeit nichts tut am Ladeport. VAG-Autos sind solche Kandidaten. 
Unter Einstellungen->Fahrzeug die "Ausstecken Simulieren" an? 10s sind standard und reichten bei mir für den Citigo und Enyaq aus.

gyroslav, 2023-07-07:
Zur Überschrift: "car zeigt status 4 obwohl auto nicht voll geladen" 

CAR=4 heißt nicht, dass das Auto voll ist. Es zeigt nur an, dass die Ladung angehalten wurde. 
CAR=3 ist der sehr kurze Status nach einer Neuverbindung und bevor die erste Ladung losgeht.