goecharger/go-eCharger-API-v2 #163: Modbus Datenformat

rainer4, 2023-07-18:
Beim Lesen der Modbus Register POWER_Lx und POWER_TOTAL erhalte ich andere Formate
146 147 | POWER_L1 | Input Register | uint32_t | 2 | Leistung auf L1 in 0.1**kW** (36 entspricht 3, 6**kW**)
z.B. bei 2500W auf POWER_L1 erhalte ich den Wert 25000 ein ==> d.h. Leistung auf L1 in 0.1**W** (36 entspricht 3,6**W**)

120 121 | POWER_TOTAL | Input Register | uint32_t | 2 | Leistung gesamt in 0.01**kW** (360 entspricht 3, 6**kW**)
z.B. bei 6000W Ladeleistung erhalte ich den Wert 600000 ==> d.h. Leistung gesamt in 0.01**W** (360 entspricht 3,6**W**)

Firmware 055.5 HW V3

0xFEEDC0DE64, 2023-07-23:
danke für den Hinweis, ich werde die Doku dann morgen aktualisieren!

rainer4, 2023-08-19:
Ich sehe die Korrektur noch nicht? Aber es wäre doch besser die Auflösung zu ändern 0.1W bei POWER_Lx bzw. 0.01W bei POWER_TOTAL ist doch übertrieben.  1W bei POWER_Lx und 1W bei POWER_TOTAL wäre doch viel besser, VOLT_Lx hat doch auch 1V Auflösung.

MYG63, 2023-10-20:
Wäre mal interessant was hier Sache ist. Ist tatsächlich der Wert jetzt Faktor 1000 oder nicht? Wir haben hier gerade eine V1 und eine V2 und die Werte sind Faktor 1000 unterschiedlich. Bitte um Klärung. Danke

rainer4, 2024-07-07:
Das Register POWER_TOTAL wurde in der aktuellen Doku korrigiert, POWER_Lx ist in der Doku immer noch falsch angegeben.
Nach wie vor finde ich die Auflösung von 0.1W bei POWER_Lx bzw. 0.01W bei POWER_TOTAL übertrieben.
1W bei POWER_Lx und 1W bei POWER_TOTAL, würde dann auch zu VOLT_Lx mit 1V Auflösung passen.