goecharger/go-eCharger-API-v2 #171: ECO-Modus schaltet nicht ab

wrzlbrmft1, 2023-08-20:
Bei mir schaltet der ECOModus nicht ab, wenn kein Überschuss vorhanden ist.
Folgende Werte sind gesetzt:
- psm = 0
- lmo = 4
- fup = true
über MQTT wird im Abstand  von 5s folgendes gesetzt:
- Payload: {"pGrid":8045.713791088695,"pPv":0,"pAkku":0}
- Topic: go-eCharger/xxxxxx/ids/set
Die Ladeleistung (Topic: go-eCharger/xxxxxx/tpa = 5214
In der App wird "Eco" angezeigt. Normalerweise müsste doch bei diesen Werten das Laden beendet werden.

Leider wird die Stromstärke auf 6 A reduziert, die Phasen auf 1 reduziert, aber die Ladung nicht beendet.

- Gibt es eine Reihenfolge, in welcher die Werte psm, fup und lmo gesetzt werden müssen?

gyroslav, 2023-12-08:
was sagt denn der Key "rfc"?

wrzlbrmft1, 2024-05-22:
> was sagt denn der Key "frc"?

Hallo - ich wollte das PV-Laden wieder aufgreifen. Aktuell funktioniert die Abschaltung am Abend wieder nicht.

Das Datenpaket schaut folgendermaßen aus:

- /ids/set: {"pGrid":1134,"pPv":515,"pAkku":1138}
- /frc = 0
- /fup = true
- /psm = 0
- /lmo = 4

Trotzdem wird mit niedrigster Leistung (1-phasig, 6A) geladen

wrzlbrmft1, 2024-05-22:
. . .  hab noch was entdeckt:

Der modelStatus steht auf 20 (ChargingBecauseChargePauseNotAllowed=20,)

erxbout, 2024-07-10:
Gleiches problem wie in #226

Der api-key "acp" muss gesetzt werden..

Auszug aus der doku:

| Key        | R/W        | Type                         | Category      | Description                                                                         |
| ---------- | ---------- | ---------------------------- | ------------- | ----------------------------------------------------------------------------------- |
| acp        | R/W        | bool                         | Config        | allowChargePause (car compatiblity)                                                 |

closed 2024-10-15 by erxbout