goecharger/go-eCharger-API-v2 #173: Fehlende Daten ModBus

tops4u, 2023-08-23:
In der Vergangenheit wurden einige Keys in ModBus aufgenommen was ich toll finde. Leider fehlen mir noch 2 weitere:

_ModelStatus_ -> Bei einer Phasenumschaltung verändert sich CAR. Um nun fest zu stellen ob dies Änderung aufgrund einer Phasenumschaltung stattfindet, werte ich Modelstatus aus. Leider ist dieser Key auf ModBus (noch) nicht vorhanden.

_DWO_ -> Da ich unser Auto in der Regel auf 80% Lade (Akku schonend) setze ich automatisch bei jedem Ladevorgang DWO. Auch dieser Key ist im Moment auf Modbus nicht vorhanden. 

Zur Zeit nutze ich Modbus und MQTT möchte jedoch vollständig auf ModBus wechseln.

0xFEEDC0DE64, 2023-10-02:
Ich bau die fehlenden Parameter gerne in unser Modbus Profil ein!

Aber mich würde mal interessieren, warum bei Modbus so viel Interesse da ist...
das Protokoll ist alt und hässlich, macht es einem schwer normale Datentypen wie int32 oder float/double oder gar Strings zu verwenden,
ist nicht selbst beschreibend mit topics oder api key namen, stattdessen aber zufällige Ziffern in der Adresse.
Jeder Hersteller macht die Byte- und Register-Reihenfolge anders.
und Mechanismen, um mitzuteilen, welcher Wert genau sich jetzt geändert hat (delta-updates) gibts auch nicht, man muss als Modbus-Master immer alles pollen und so maximal Traffic verursachen.

tops4u, 2023-10-05:
Ich verwende gerne MQTT für kleine IoT Geschichten (auch selbst gebaute), mit einer übersichtlichen Anzahl an Parametern. Bei der Wallbox bin ich etwas erschlagen von der schieren Menge (und den kryptischen Bezeichnungen) an Parametern das macht den Transfer aufwändiger und ergibt mir viel mehr Datenpunkte in IOBroker als ich eigentlich haben möchte. Da habe ich es lieber gezielt die Daten abzuholen, welche ich brauche. Da meine Wechselrichter eh auch nur ModBus Sprechen ist das Handling weniger ein Problem.

Über 220 Datenpunkte werden nur durch die MQTT Anbindung von go.e hinzugefügt. Für meine Zwecke brauche ich jedoch eigentlich nur 12 (davon sind 3 doppelt, weil ich die auch Schreiben möchte).

0xFEEDC0DE64, 2023-10-25:
DWO oder energy_limit ist schon seit einigen Monaten im Modbus Profil mit dabei, siehe register 40334 (333) ENERGY_LIMIT in der [modbus doku](https://github.com/goecharger/go-eCharger-API-v2/blob/main/modbus-de.md),

was noch über bleibt ist der modelStatus

rainer4, 2024-07-07:
> Ich bau die fehlenden Parameter gerne in unser Modbus Profil ein!
>
> Aber mich würde mal interessieren, warum bei Modbus so viel Interesse da ist... das Protokoll ist alt und hässlich, macht es einem [...]

Es gibt halt noch Hardware wie z.B. eine Siemens Logo 8.3 welche nur Modbus unterstützt (nur die aktuellste Logo 8.4 kann MQTT), daher wäre es schön wenn go-e den Modbus weiterhin unterstützt.