goecharger/go-eCharger-API-v2 #175: Negativer Ladewert

tdxchris, 2023-08-28:
Ich erhalte mit meinem Go-E Charger teilweise negative Ladewerte. Hat das schon mal jemand gesehen oder eine Lösung dafür?
[image]

fdmarder, 2023-09-01:
tritt bei mir nicht auf, bzw. habs noch nie gesehen.

menkej, 2024-02-11:
Habe ich bei mir auch regemäßig, beim Start der Ladung meines Aiways U5. Führt dann immer zu einer Warnung in der Steuerung (EVCC). Hat irgendwie ein Geschmäckle, aber bisher habe ich - bis auf die Warnung - nichts negatives feststellen können.

cgenser, 2024-04-03:
Tritt bei mir ebenfalls in den ersten Sekunden nach Start der Ladung auf.

Führt zu Problemen, wenn man die Leistung per Modbus ausliest, weil POWER_TOTAL (auch POWER_L1, POWER_L2, POWER_L3) laut Doku ein uint32_t ist, aber vermutlich int32_t sein sollte.