goecharger/go-eCharger-API-v2 #177: Wrong time values?

enso-media, 2023-09-22:
Can anyone explain values like lccfi, lccfc and lcctc?
Based on the documentation i expected those parameters to be timestamps that can be used to get the start and stop times of the latest charge or state change.
Its documented as "ms", but the values i get from the API makes no sense at all. They dont change after a charge, so its not a difference. Also i get values like "179773240" which would be like 2 days. Its also not a valid unix like timestamp.

0xFEEDC0DE64, 2023-10-02:
those timestamp api keys take the reboot time (api key rbt).

the idea was that you subtract rbt - lccfi for example, and you get a relative timespan when the event happended or will happen.

when you change the utc or local time, or when NTP time sync happens, or you switch timezones, then the reboot time is not affected as its just the time in milliseconds since the device was powered on, it is a monotonic clock, unlike the utc clock for example.

karliwalti, 2024-05-24:
Is this also true for lmsc, inva, and lpsc ?