goecharger/go-eCharger-API-v2 #178: Charging Session Export erwartet zusätzlichen Klick

Chirriu, 2023-09-26:
Ich habe seit einiger Zeit per Script meine Charging Sessions einmal täglich exportiert und in eine Datenbank geschrieben zur Sicherung, falls die Wallbox mal das Zeitliche segnet. 

Über den status call hole ich mir dir URL unter "dll" und konnte per einfachem urlopen die Daten abspeichern. 

Wie es scheint, gab es aber nun eine Änderung, die einen zusätzlichen Klick erfordert und damit natürlich die URL praktisch so nicht mehr zu verwenden ist. 

Kann das wieder lauffähig gemacht werden, sodass die Exporte auch programmatisch vorgenommen werden können? Was ist die Alternative?

Chirriu, 2023-10-02:
Hi, 
sehr cool, vielen Dank. 
Das erste Mal durchlaufen lassen lief nach ca. 45 Minuten in einen 504 Timeout.

0xFEEDC0DE64, 2023-10-05:
der log export für dich war nach einer minute fertig aber die response wurde leider nicht beendet, sollte nun gefixt sein!

Chirriu, 2023-10-05:
Ich hab es gerade laufen lassen und mein Script angepasst. Alles läuft wieder wunderbar. Sehr sehr geil. Vielen Dank für den Support.

closed 2023-10-05 by Chirriu