goecharger/go-eCharger-API-v2 #180: MQTT Sende Interval einstellbar machen.

juergen852, 2023-10-02:
Da über MQTT jede Sekunde die Veränderungen z.B. der Stromspannung, Stromfrequenz, rbt, rssi etc.geschickt wird, entsteht da viel unnötiger Traffic.

Es hat ja nicht jeder nur einen Go-E Charger am laufen....

Da die meisten Werte auch alle 10, 60 oder 600 Sekunden ausreichen würden, wäre es hilfreich, wenn man einen MQTT Sende-Rhythmus einstellen könnte.

Meinetwegen sogar 2 Timer, einer während des Ladevorgangs und einer während kein Ladestrom fließt.

Jede Sekunde den MQTT-Server und den Server, der die MQTT-Infos danach aufbereitet zu "beschießen" erzeugt dort eine unnötig hohe Last ohne Mehrwert.

0xFEEDC0DE64, 2023-10-25:
Ein einstellbarer Intervall ist geplant, und ausserdem wollen wir die Menge an api keys einschränkbar machen, zb man definiert einen filter wie bei der http api, welche api keys durch kommen und welche nicht. das würde auch einiges an Last am Microcontroller einsparen und die box so mehr responsive machen

arteck, 2024-01-17:
habt ihr da ne roadmap für ??

juergen852, 2024-01-31:
Bitte schaut bald mal danach, der "MQTT spam" ist echt "hinderlich". 
Auch auf dem Go-E Charger erzeugt das ja ziemlich Last.

naseweiss99, 2024-06-30:
hat sich hier was getan ? - das ist viel zu oft, ich sehe im Log der Hausautomation nur unnötige MQTT Telegramme vom GO-e.

rhuss, 2024-07-31:
Waere auch an einer solchen Option interessiert, und das schon seit 4 Jahren --> https://github.com/goecharger/go-eCharger-API-v1/issues/5

juergen852, 2024-09-16:
@0xFEEDC0DE64  Frage:

Bringt es überhaupt etwas, hier Fehler oder Wünsche zu melden?

Auf welchem Weg soll man Go-E helfen, ein brauchbares Produkt hin zu bekommen?

Wenn ich mir die Issues hier anschaue, scheint das eher ein Hobbyprojekt zu sein....

schinagl, 2024-09-17:
> 
> Wenn ich mir die Issues hier anschaue, scheint das eher ein Hobbyprojekt zu sein....
> 

Schlecht geschlafen?
Es zahlt sich immer aus etwas konstruktiv beizutragen. 

Welcher andere Wallbox kann Überschuss, Phasenumschaltung und ist so von außen steuerbar, uvm um diesen Preis?

Finde das und den Support bei einem anderen Hersteller.

erxbout, 2024-10-01:
Leider haben wir begrenzt Entwickler Ressourcen zur Verfügung und viele offene Baustellen..

Wäre das Verwenden von OCPP denkbar? Laut der Webseite von symcon.de sind die damit kompatibel..

Grundsätzlich würde ich schon sagen, dass hier der Ort für solche fragen ist.

juergen852, 2024-11-18:
Gerade getestet, FW56.8: 
Bei OCPP ist der Intervall einstellbar.
Bei MQTT derzeit kein Intervall konfigurierbar, sobald sich z.B. die Netzfrequenz ändert, werden im Sekundentakt alle Werte geschickt.

halmich, 2025-03-06:
Ich warte ebenfalls sehnsüchtig, dass go-e das mqtt Sendeintervall einstellbar macht... 

@schinagl 
> Welcher andere Wallbox kann Überschuss, Phasenumschaltung und ist so von außen steuerbar, uvm um diesen Preis?

Zappy?

juergen852, 2025-04-06:
@erxbout Go-E scheint die ID der RFID-Chips nicht in den Daten mitzuliefern.
Zumindest konnte ich in den Datensätzen, welche bei Symcon per OCPP ankommen nichts davon finden.
- Deshalb lässt sich die Go-E auch nicht nach "Nutzerkontrolle" in Symcon zum landen freischalten.
- Der Go-E ist, wenn OCPP aktiviert ist, auch nicht mehr an der Ladestation per Chip freischaltbar, da muss per OCPP freigeschaltet werden, was natürlich nur nach Identifikation per Chip passieren darf....

Nachstehend, was da per OCPP ankommt.
(Ich hatte natürlich im Debug auch die Rohdaten untersucht, kein Chip drin.... Die Transaction-ID hat ja nichts mit dem Freischalt-Chip zu tun....
[image]

rhuss, 2025-08-19:
> Ein einstellbarer Intervall ist geplant, und ausserdem wollen wir die Menge an api keys einschränkbar machen, zb man definiert einen filter wie bei der http api, welche api keys durch kommen und [...]

Gibt's schon ein Update gegenueber dem Stand von Oktober 2023 ? Auch ein: "Nein, werden wir nicht implementieren" waere für persoenliche Planungen hilfreich (noch hoffe ich ja, dass das geht, da ich es mir eigentlich nicht so schwierig vorstelle das konfigurierbar zu machen. Diejenigen, die implizit auf Sekunden Intervalle sich verlassen (um z.b. eigenen Leistungswerte zu berechnen) muessen den default ja nicht aendern)

codeproducer198, 2026-10-17:
Da ich den Charger Gemini lokal (ohne Cloud) nutze und auch lokal eine verschlüsselte Verbindung will, kann ich nur auf MQTT/TLS setzen. Und da sind die unnötigen sekündlichen Updates vom Charger nervig - vor allem wenn das Auto nicht angeschlossen ist/geladen wird.
Deshalb **bitte bitte bitte diese Funktion umsetzen**, so kann ich Last von meinem MQTT, Home Assistant und DB nehme.

Habt ihr schon einen groben Zeitplan für uns Kunden/Nutzer?

chk2902, 2026-10-17:
"Da ich den Charger Gemini lokal (ohne Cloud) nutze und auch lokal eine verschlüsselte Verbindung will, kann ich nur auf MQTT/TLS setzen"

Das geht aber auch über die WebSocket-Verbindung. Die nur m.W. nicht von go-e dokumentiert ist, aber Implementationen findet man bei github und meine App nutzt die auch, wenn man das möchte. Rein interessehalber: wieso brauchst Du das lokal verschlüsselt? Das Prinzip sehe ich ein, nur nicht, wieso jemand, der in Dein WLAN eingebrochen ist, sich gerade dafür interessieren sollte.

Grübel... ich muss meine App mal zu einem go-e WebSocket-zu-MQTT Proxy umbauen ;)

codeproducer198, 2026-10-17:
> "Da ich den Charger Gemini lokal (ohne Cloud) nutze und auch lokal eine verschlüsselte Verbindung will, kann ich nur auf MQTT/TLS setzen"
>          Das geht aber auch über die WebSocket-Verbindung. [...]

Ich bin da ein bischen Paranoid. Deshalb ist bei mir alles was möglich ist, auch "intern", verschlüsselt/TLS. Und auf meinem lokalen Debian Server, wo neben vielen anderen Services auch MQTT läuft, will ich halt keinen Klartext Traffic haben.

@chk2902
Zu Websockets: Meinst du OCPP via wss? Habe ich mir tatsächlich noch nicht angeschaut. Funktioniert das Gemini da mit Home Assistant Integrationen vernünftig?

chk2902, 2026-10-17:
Nein, es gibt eine API, die verschlüsselte Pakete mit dem go-e austauscht (Auth und Wallbox-Kommandos direkt zu senden, Informationen kommen anfangs einmal voll, dann in kleinen Delta-Paketen per UDP sekündlich). 

HomeAssistant hat die API anscheinend nicht drin (meint Gemini).

Siehe auch **https://github.com/goecharger/go-eCharger-API-v2/issues/87**, da ist es das fullStatus-Paket.

OCPP hatte ich mit meinem Core mal probiert (um OCPP in meiner App anbieten zu können), das lief, zumindest bei meinen Funktionstests. Ist mir aber eigentlich "nur" zur Steuerung der Wallbox unnötig komplex, dass man da einen OCPP-Server anbeiten muss. Zumal OCPP derart fehleranfällig ist, wenn man hier mal die OCPP-Issues anschaut... ich weiß nicht. Nett gemeint, aber für Einzelinstallationen wie Kanonen auf Spatzen (sorry, ihr süßen Viehcherlein).

Dammit - meine App könnte tatsächlich genau das machen, sie kann diese Websocket-Kommunikation und auch MQTT. Andererseits kannst Du mit über MQTT die App und damit die Wallbox steuern (https://forum.chk.digital/viewtopic.php?t=353), aber dazu musst Du ein Android-Smartphone oder Tablet dauernd neben her laufen lassen - wohl eher etwas Overkill für Deine Zwecke.

chk2902, 2026-10-17:
Was mir gerade aufällt: die haben die "Delta"-Information also tatsächlich auch schon in der Software, d.h. sie wissen, welche Datenpunkte sich tatsächlich ändern und welche sie nur schicken müssen. Denn nur die schicken die ja in den "delta"-Statuspaketen.

naseweiss99, 2026-10-17:
es geht ja eigentlich um die Frequenz der MQTT Aussendungen des GO-E.
Ich kann momentan Keinem mit einer Heimautomation und MQTT den GO-E empfehlen. 
Nächstes Jahr 2027 benötige ich eine weitere Wallbox - ich werde mir keine GO-E kaufen, wenn die MQTT Intervalle bis dahin nicht einstellbar sind.
Ich habe den Eindruck, in der GO-E Software Entwicklung geht denen das kilometerweit am Gesäß vorbei.

chk2902, 2026-10-17:
Klar, aber so kämst Du ohne MQTT zurecht. Und ja, ich glaube MQTT ist so ein "Häkchen"-Feature, das manche Software so hat, damit das in den "unterstützen Schnittstellen" so auftaucht. 

Wenn der HA OCPP unterstützt, kannst Du es mal damit probieren. Dann sind auch mehrere Wallboxen kein Problem.

Edit: ach so, anderer User. Wie machst Du die Ansteuerung?

marq24, 2026-10-27:
Mahlzeit!

Eigentlich wollte ich mich hier gar nicht in diese Diskussion einmischen. Denn die grundsätzliche Lösung für dieses „Problem“ ist aus meiner Sicht trivial:

> _Wenn Dich etwas nicht interessiert, dann höhr‘ einfach nicht zu – bzw. wenn es Dich nicht interessiert, dann verarbeite es auch nicht._

Genau dieser Grundsatz trifft zumindest für mich auf dieses komplette Thema zu.

Ich bin glücklich damit das mich ein Hersteller mit einem konstanten Datenstrom versorgt, und zwar genau dann, wenn sich diese Daten auf dem Gerät auch ändern. Es liegt an mir (dem Leser), ob ich diesen Daten verarbeiten möchte/kann oder eben nicht.

Zumindest mein go-e zwingt mich nicht, sich per MQTT zu subscriben. Es gibt zwei weitere Möglichkeiten an die Daten der Wallbox zu gelangen – per WebSocket oder eben die REST-API (und das alles lokal - ganz **ohne** Cloud-Zwang!).

Andere Hardwarehersteller sind da wesentlich beschränkter, was den Zugang zu meinen Daten meiner eigenen Hardware angeht (allem voran die Fahrzeughersteller – ganz besonders VW, oder auch AUDI).

Ihr könnt mich gerne als go-eFanBoy beschimpfen – für mich als Entwickler sind die Möglichkeiten die mir go-e bietet einfach optimal. Allein schon die Existenz dieses öffentliche Repos zeigt so viel Transparenz und Offenheit, wie ich sie von _keinem anderen kommerziellen Hersteller_ (dessen Hardware ich besitze) her kenne. 

Um aber auch noch eine mögliche Lösung (zumindest für einen Teil von Euch) zu teilen: Für _Home Assistant_ gibt es eine Integration (‚go-e APIv2 Connect ‘ über HACS) mit der die Daten über die Websocket Verbindung vom go-eCharger zwar sofort gelesen werden – aber die eigentliche Aktualisierung der Entitäten (also die Verarbeitung) kann gedrosselt werden. Diese wäre aus meiner Sicht zumindest ein Weg einen solchen „Filter“ zu realisieren.

juergen852, 2026-10-27:
@marq24 
Nicht subscriben.... Was für ein toller Tip !

Natürlich will ich zuhören, sonst würde ich nicht subscriben. 
Wenn ich aber nicht mehr zuhören kann, weil die Lautstärke mir die Ohren kaputt macht, dann ist das ein Problem.
Wenn ich wirklich jede Änderung will, dann nehme ich Websockets und nicht MQTT.

Ich kenne viele Geräte die per MQTT senden und bei denen ist das alles kein Problem, weil vernünftig gemacht.

Go-E Denke:
Eine Frequenzänderung um 0,01 Hz. ist eben kein Trigger für den Versand aller Variablen.
Ebenso brauch ich auch nicht bei jedem 0,5 Volt das sich ändert den kompletten Datensatz.

Realität:
Die Hardware kann die Messung ja nicht mal mit der Genauigkeit durchführen, deshalb wäre es angebracht hier erst bei 0,5Hz oder 2 Volt Änderungen zu senden.

Ebenso macht es keinen Sinn, alle 100 Millisekunden was zu senden.
Das ist keine Echtzeit-Anwendung, sondern ein Ladegerät....
Da reicht ein Update alle 5 Sekunden, da die Welt in der Zwischenzeit nicht untergeht.

In der Go-E App wird ja auch nicht in Millisekunden aktualisiert, weshalb sollte das bei MQTT richtiger sein?

marq24, 2026-10-27:
Entschuldig @juergen852 - es scheint als hätte ich mich hier missverständlich ausgedrückt (bzw mein eigentliches Anliegen ist nicht deutlich geworden). Ich wollte sagen sagen, dass es neben dem Raum [in dem es Dir zu Laut ist (und von dem Du sagt, das die Lautstärke aus Deiner Sicht überhaupt keinen Sinn ergibt)] noch weitere Räume von go-e bereitgestellt werden in dem Dir die Lautstärke ggf. mehr zusagt - bzw diese sogar regelbar ist (über Umwege). Aber 'ja' in diesen alternativen Räumen wird nicht MQTT gespielt.

Mir fehlt der technische Hintergrund, um mir ein Urteil zu erlauben, ob die Update-Frequenz der MQTT Nachrichten jetzt "sinnvoller/weniger sinnvoll" ist als bei anderen Herstellern.

> Da reicht ein Update alle 5 Sekunden, 

Das ist eine Meinung - mir wären 5 Sekunden das schon zu langsam... auch wenn ich Dir im Grundsatz zustimme, das die Welt in 5 Sekunden nicht untergehen wird.

> In der Go-E App wird ja auch nicht in Millisekunden aktualisiert, 

Sobald über die App über die WebSocket connection neue Daten in der App empfängt, werden diese auch umgehen angezeigt. Da gibt zeitlich kein Delay. Wenn Du z.B. über die REST-API einen Wert schreibst, siehst Du auch unmittelbar in der App wie sich der Wert/Setting ändert. Das ist aber auch alles eigentlich off-topic - sorry 

Meine Aussage ist, "neben MQTT" gibt es noch zwei weiter alternative Kanäle die einem bei go-e offen stehen... Wenn Dir die go-e MQTT-Musik "zu laut" ist, dann habe ich eine alternative im Home Assistant Universum vorgeschlagen, die dieses Problem lösen könnte.

Wenn es aber unbedingt MQTT sein muss, könntest Du z.B. via Node-RED oder einem einfachen Python Script einen Custom MQTT Proxy dazwischen hängen der die Pufferung für Dich übernimmt - und die Lautstärke auf ein für Dich erträgliches Maß herunter regelt - Claude, ChatGPT oder auch Gemini sind da bestimmt hilfreich.

juergen852, 2026-10-27:
@marq24 : Sorry, wenn ich da emotional wurde.

Das was Go-E in dem Thema macht ist einfach nicht professionell.

Weißt Du, ich hab nicht einen sondern 32 Stück Go-E dann wird MQTT laut.
Workarounds mit Proxies oder sonstigem Kram: Ich hab schlicht keinen Bock dazu, die Entwickler-Shortcomings auszubügeln.
Anderes Protokoll: Siehe u.a. in diesem Thread die Kommentare zur Stabilität von OCPP. 
Das ist bei den anderen Schnittstellen von Go-E leider nicht besser.

Vielleicht sollten die Go-E Entwickler mal Deinem Tip mit " Claude, ChatGPT " folgen.....
Jeder Tasmota kann "teleperiod", aber für Go-E scheint das Magie zu sein.

Und wenn Entwickler nur auf Kleinkunden hören, dann ist das was: ?

Nachdem ich Go-E jetzt über 2 Jahre Zeit gegeben habe:
Die Beschaffung weiterer Go-E ist bei uns gestoppt.
Diese Woche 10 Geräte eines anderen Herstellers im Zulauf.

marq24, 2026-10-27:
@juergen852 bei 32 go-e kommt sicherlich einiges an MQTT-Messages zusammen - damit wird ein Pie sicherlich nicht mehr mit fertig... da wird die Verwaltung und Skalierung eines solchen Systems sicherlich erhöhte Anforderungen an Netzwerk, Broker-Performance und Themenstruktur haben - um nicht zu sagen, da ist MQTT vielleicht dann doch einfach ungeeignet.

> Anderes Protokoll: Siehe u.a. in diesem Thread die Kommentare zur Stabilität von OCPP.

Von OCPP habe ich gar nicht gesprochen (bzw, ich habe damit Zero-Erfahrung) auch wenn ich verstehe, das das eigentlich "the way to go" bei dem Betrieb eines (kleine) Ladeparks sein sollte.

Die WebSocket Option ist für meinen persönlichen UseCase optimal (aber ich bin eben auch nur ein _Kleinkunde_) - und über die Home Assistant Integration kann ich alles so steuern wie ich das in meinem privaten Umfeld möchte. Man _muss_ eben nicht MQTT verwenden um seine 1-3 Wallbox(en) steuern zu können.

Ob ich das ganze allerdings für einen 24/7 Betrieb von 32 Ladestellen einsetzen möchte? Sagen wir mal _Ich weis nicht_. Das sprengt dann doch den Rahmen einer Heimanwendung. Ggf muss man doch etwas tiefer in die Tasche greifen und andere Kommerzielle-Lösungen installieren. Wenn ich Dich richtig verstanden habe, dann machst Du dies jetzt auch - viel Glück (bzw hoffentlich halten die Datenblätter ihre versprechen).

Es ist sicherlich auch nicht optimal, wenn es eine Rückmeldung hier auf GitHub gibt "_Wir arbeiten dran..._" dann aber (aus welchen Gründen auch immer) für Dich als Endanwender nicht wirklich etwas erkennbares passiert (und das über n-Jahre!) - Ja das erzeugt am Ende Frust. Vielleicht kann man das dann aber auch unter "_aus Fehlern für die Zukunft Lernen_" verbuchen? 

Denn auf der anderen Seite ist go-e der einzige Hersteller in meinem privaten Hardware Universum, der überhaupt so offen und transparent bzgl. seiner APIs und der Weiterentwicklung ist. Natürlich kenne ich nur _meine_ Hardware. Aber für ausnahmslos alle anderen Geräte im Haus (Wärmepumpe, PV-Speicher, EV) musste ich die Schnittstellen mir mühsam selbst erschließen und lokal nutzbar machen. Deswegen wünschte ich mir es gäbe mehr Hersteller, die den Weg von go-e folgen, auch wenn dies bedeutet, das nicht immer alles umgesetzt werden kann, was in einem Issue versprochen wurde.

chk2902, 2026-10-28:
... und es ist doch *so einfach* zu lösen, denn wie gesagt wissen die prinzipiell schon, was sich alles an Daten ändert. Klingt ein bisserl nach - Vorsicht Verschwörungstheorie - "never change a running system". Kenne ich von Microsoft, die schon vor 20 Jahren nicht mehr an Code ranwollten, wenn man ihnen einen Bug gemeldet hat.

Könnte auch sein, dass denen die Entwickler ausgehen. Das kenne ich auch von meiner Haupt-Arbeit, wer kennt sich heutzutage noch mit Embedded Systemen und deren Software(beschränkungen) aus? Also außer im Einsatz in Hobbyprojekten?

Ich hoffe trotzdem, dass dem nicht so ist. Denn im Prinzip mag ich die go-e sehr.

juergen852, 2026-11-01:
@erxbout 
Update erbeten

erxbout, 2026-11-03:
@juergen852 Kann euch nicht helfen, arbeite dort nicht mehr und würde die geforderten Änderungen selbst begrüßen..

chk2902, 2026-11-03:
Dann lag ich nicht wirklich weit weg von der Erklärung... schade. Dann müssten die die neue Websocket-Schnittstelle dokumentieren, dann ist auch wieder alles gut auch für Dutzende von Boxen. 

MQTT ist halt das Sorgenkind - daher habe ich in meiner App MQTT-Unterstützung auch wieder rausgeworfen. In jeder Familie gibt es ein schwarzes Schaf.

juergen852, 2026-11-03:
@erxbout Danke für die Info.

0xFEEDC0DE64, 2026-11-05:
Habe gerade einen neuen api key "mpi" implementiert: MQTT publish interval in seconds.
Kommt mit dem nächsten releasten firmware update mit

codeproducer198, 2026-11-08:
@0xFEEDC0DE64  tausend Dank Daniel. Kanns kaum erwarten das einzustellen/auszuprobieren. Ist schon bekannt, wann das nächste Release (Beta?) freigegeben wird?