goecharger/go-eCharger-API-v2 #181: Ladeleistung in Watt per MQTT vorgeben

DanielWeigl, 2023-10-04:
Hallo,

ich steuere meine goE per Homeassistant via MQTT, momentan setze ich den Ladestrom via `/amp` ApiKey, mit dem ist es aber nur sehr grob (in ganzzahigen Amper-Schritten ist das bei 3-Phasen ja um die 700W) und ich muss mich auch selber um die 1phasen/3phasen Umschaltung kümmern, wenn ich unter ~4kw will.

Hab ich da was übersehen - oder wäre es möglich einen eigenen ApiKey zu bauen bei dem man die maximalen Watt vorgebem kann und die Wallbox macht den Rest selber?

povlhp, 2023-12-05:
You need to use the pv functionality. It is used for charging with surplus energy. I run this every 5 seconds (needs at least once per 10s)
Basically you only need to set pGrid.  If it is negative, it means you are exporting energy, and the charget will try to use only that amount.

So set pGrid to -3900, and the Go-e will try to charge with 3900W in in "use pv surplus" mode.

The standard only allows for whole amps, so there is nothing like 6.3A on 3 phases. It is 6A or 7A. I asked if we could get an option to allow continous values. Then we could figure out if the car would round off or use the value as 6.3A.

```
alias: Go-echarger update solar
description: 
trigger:
  - platform: time_pattern
    seconds: /5
condition: []
action:
  - service: mqtt.publish
    data_template:
      topic: /go-eCharger/055550/ids/set
      payload: >-
        {{'{"pGrid": '}}{{states('sensor.power_total')}}{{', "pPv":
        '}}{{states('sensor.solutronic')}}{{', "pAkku":0}'}}
    enabled: true
mode: single
```

BaaZinga23, 2024-02-09:
pGrid is only a readable value? I search a solution for the same problem.

tennox, 2024-03-13:
@BaaZinga23 I've read in a few places (e.g. https://nasauber.de/blog/2023/go-e-charger-und-pv-ueberschussladen-update/) it's only documented as read-only, but actually writable. Opened an issue about that https://github.com/goecharger/go-eCharger-API-v2/issues/199