goecharger/go-eCharger-API-v2 #188: What is "cll" - a new API key not yet documented except in the firmware version history

chk2902, 2023-12-31:
"cll" is "aktuelle Strom Limits". OK, was ist das?

0xFEEDC0DE64, 2024-02-01:
current limits list,

ein json object, das jedes berücksichtigte limit in der ladelogik ausdrückt,
am ende fällt ja immer der maximale ladestrom heraus aber man wusste halt nie, wie sich der zusammensetzt.

der api key beinhaltet wirklich jedes zwischenergebnis und wird demnächst in der app benutzt.

beispielhafte bedeutung:

limitiere auf 22A wegen kunden-einstellung,
limitiere auf 21A wegen elektriker/installateur-einstellung
limitiere auf 20A wegen übertemperatur,
limitiere auf 19A wegen lastmanagement,
limitiere auf 18A wegen ocpp,
limitiere auf 17A wegen temporärem limit,
limitiere auf 16A wegen adapter erkennung,
limitiere auf 15A wegen zu niedriger netzfrequenz,
limitiere auf 14A wegen überschussregeln,
limitiere auf 13A wegen maximaler unsymetrie am 3phasigen Anschluss.

aber meistens treffen nicht sooo viele limits auf einmal zusammen :)

marq24, 2024-07-28:
@0xFEEDC0DE64 Is there a list of the possible available fields in the cll object?

I just can assume there are more then just the ones currently showing up here in my local installation?

```
         "accessControl": ..,
         "currentLimitMax": ..,
         "minChargingCurrent": ..,
         "requestedCurrent": ..,
         "temperatureCurrentLimit": ..,
         "unsymetryCurrentLimit": ..
```