goecharger/go-eCharger-API-v2 #190: Phase switching behaviour when setting charging power via OCPP

mincequi, 2024-01-12:
Hi there,
 i am actually interested how the go-echarger behaves when i provide the allowed charging power via OCPP SetChargingProfile. Does the go-echarger do phase switching then (e.g. 2000W -> 1 phase @8A and 8000W -> 3 phase @11A)? And is this somehow debounced to prevent hardware damage?

Thanks a lot!

0xFEEDC0DE64, 2024-02-01:
our OCPP stack only allows for specifying the current in A and not the power in W.
By the way the OCPP server may also specify the numberOfPhases parameter to limit the number of phases to 1

mincequi, 2024-02-01:
@0xFEEDC0DE64 
Ok, perfect. Thanks for your feedback.

Could you provide information on the device's tolerance to phase switching and any potential damage as a result of a lot of switching?
What would you consider as a sane phase switching rate?