goecharger/go-eCharger-API-v2 #198: Unerwartet: Ladestart nach Phasenumschaltung löscht "geladene Energie" ("wh")

chk2902, 2024-03-08:
`03:00:00.354 '{"alw":false,"acu":null,"adi":true,"dwo":null,"eto":2526179,"lmo":3,"ama":16,"amp":6,"ust":0,"fsp":true,"acs":0,"frc":1,"car":4,"err":0,"cbl":20,"pha":[false,false,false,true,true,true],"wh":3115.695425,...`

`03:00:00.396 api/set?amp=6&ct=%22%22&psmd=10000&psm=2`

`03:00:15.363 24694: .devices.wallbox.go_eCharger.Wallbox_Data@4224fae: {"alw":true,"acu":6,"adi":true,"dwo":null,"eto":2526179,"lmo":3,"ama":16,"amp":6,"ust":0,"fsp":false,"acs":0,"frc":0,"car":1,"err":0,"cbl":20,"pha":[false,false,false,true,true,true],"wh":3115.695425,...`

`03:00:20.486 api/set?acs=0&frc=1`

`03:00:31.260 '{"alw":false,"acu":null,"adi":true,"dwo":null,"eto":2526179,"lmo":3,"ama":16,"amp":6,"ust":0,"fsp":false,"acs":0,"frc":1,"car":4,"err":0,"cbl":20,"pha":[false,false,false,true,true,true],"wh":0.029744708,...`

chk2902, 2024-03-08:
Kann natürlich auch die Phasenumschaltung selbst sein, wenn das Löschen des Energiewerts nur auf das Pilotsignal reagiert und das nach dem absichtlichen "Trennungs"-Vorgang wieder "verbunden" wird.

IPSCoyote, 2025-12-02:
Generell ist in der 57.1 der WH nicht stabil. Er setzt sich während eines Ladevorgangs, auch ohne Unterbrechung, Änderung der Phasenzahl oder Änderung der Ladeleistung zurück!