goecharger/go-eCharger-API-v2 #199: Clarification: ppv/pgrid docs & writable

tennox, 2024-03-13:
Hey there :)
First, thanks for openly documenting & making available an API for self-controlled Home automation :medal_military: 

There is some unclarity around how to tell the charger if (/how much) PV excess energy is available via API, instead of controlling charge state "manually" (meaning: via own software outside go-e app).

I've read in some places (e.g. this [blog post](https://nasauber.de/blog/2023/go-e-charger-und-pv-ueberschussladen-update/) or [this comment](https://github.com/goecharger/go-eCharger-API-v2/issues/181#issuecomment-1840531690)) that the parameters `ppv` and `pgrid` can actually be written to and used to achieve this.
But the documentation is lacking any description and lists them as read-only.

Could you clarify?

Twiederh, 2024-04-04:
They can be written using ids/set in MQTT. You need to provide a json like this:
{ payload: { pPv: 6000, pGrid: -5000, pAkku: 0 } };

mleck28, 2024-04-05:
Hello Twiederh,

I have some problems with this. Last Year there was no problems. But now I sent the values, I see the pgrid-value in the app, but the car doesn't load.
"Laden nicht erlaubt Eco Modus Ladepause"

Did you have any idea?

Twiederh, 2024-04-05:
Hi,
da die Meldung auf deutsch ist, vermute ich, dass wir uns auf deutsch unterhalten können :-) Ich habe das gestern erst implementiert und hatte leider bisher nicht genügend Sonne, um ausführlich zu testen. Spätestens morgen sollte ich ausführlich testen können. ich lese die Werte mittels MQTT Explorer aus, was siehst Du denn da bei pgrid? Damit das Laden starten kann, müssen mindestens 1400W Überschuss da sein. Außerdem wird das Laden 3-Phasig gestartet, wenn also nicht genügend Strom von der PV verfügbar ist, wird erst mal auf einphasig umgeschaltet und dann das Laden neu gestartet.

Wichtig ist, dass pGrid als negativer Wert geschickt wird, wenn Überschuss da ist.

Ich melde mich, wenn ich weiter getestet habe.

Viele Grüße,
Thomas.

mleck28, 2024-04-05:
Hi, vielen Dank für Deine schnelle Rückmeldung. Ich habe das negative Vorzeichen von meiner alten Version nicht mit übernommen. Daran lag es.
Jetzt habe ich das Problem das "car" nicht in Status 1 geht, wenn kein Auto angesteckt ist, stattdessen kommt eine "4". Das ist hinderlich :-(

Twiederh, 2024-04-06:
Also ich habe das heute getestet, bis darauf, dass ich in meiner Logik in Nodered einen Fehler hatte, lief das Laden einwandfrei, es ist auch wohl nicht mehr so, dass am Anfang dreiphasig geladen wird, das Laden hat bei mir einphasig angefangen und wurde auf dreiphasig umgestellt. Ich muss noch an der Ladelogik arbeiten, sodass ich das Laden bei 80% stoppen kann aber das kann ich erst machen, wenn unsere Zoe wieder leergefahren ist :-)

mleck28, 2024-04-06:
Super. Bei mir klappt es nun eigentlich auch - es lag an dem Negativwert.
Jetzt stehe ich vor dem Problem, dass ich 3 Wallboxen und zwei E-Autos habe. Ich ordne die Autos bisher den Wallboxen zu (Reihenfolge des Ansteckens). Nun muss ich aber eine Art Regler entwerfen, damit ich die Einspeisung/ Bezug für jede Wallbox bzw. dessen zugeordneten Autos berechnen und anpassen kann.

Twiederh, 2024-04-08:
Ich ich habe zwei E-Autos, aber nur eine Go-E (aber auch noch zwei Tesla Wall Connector Gen 3). Ich habe mir für die GUI der Go-E in Nodered eine Dropdown-Liste gebaut, mit der ich einfach zwischen den beiden Autos (und einem Gastfahrzeug) umschalten kann. Damit beeinflusse ich die automatische Abschaltung bei einem bestimmten SoC, bei der Zoe soll abgeschaltet werden, beim Tesla nicht (der kann das intern). Das gleiche kann man dann natürlich auch auf die Ladeleistungen anwenden.

bruma59, 2024-05-03:
what are the parameters you're using for PV surplus charging
Normally it has only to start when the minimum grid-feedin is reached, the fst value
Its start in my go-e 56.2 beta when grid-feedin is lower than the pgt value (deltap becomes negative)
when it's starts charging it keeps the power around the pgt value what's correct, 
the deltap value (positive) switches the charging off, but it kick back in with a grid-feedin value of pgt which is to low to do PV surplus, it has to wait for the fst value
lmo=4; fup=true; grid value is send in pGrid within ids (pAkku:0 and pPv:0)

Twiederh, 2024-05-03:
I do not work with the internal settings too much. I set pgt to -200 as recommended in the app itself, other than that I only send pPV, pGrid and pAkku. Since I have a home battery (Tesla Powerwall), I usually deduct 4600W from pGrid until my Powerwall reaches 90% to give it a priority, but this is configurable. Until the Powerwall is full I send pAkku with the value of 0.

bruma59, 2024-05-03:
I'm setting the parameters within the app, pgt -450, single fase, prefer feed-in, 3f switching 7kW
whatever I enter as starting power level, charging starts when PV surplus reaches (feedin level)  450W

tennox, 2024-05-29:
> They can be written using ids/set in MQTT. You need to provide a json like this: { payload: { pPv: 6000, pGrid: -5000, pAkku: 0 } };
> 
> with MQTT the endpoint is `GoEcharger/<serialnumber>/ids/set`

I configured my go-e charger to be on mqtt topic `go-eCharger`, and receive `go-eCharger/pgrid = null`.
When I do:
```
$ mosquitto_pub -h 192.168.178.147 -t go-eCharger/ids/set -m '{"pGrid": 1000, "pPv": 2000, "pAkku":0}'
```

I get a `result = unknown api key: ids`.
 
Also I tried:
- `go-eCharger/pGrid/set -m '1000'` => `unknown api key: pGrid`
- `go-eCharger/pgrid/set -m '1000'` => **`api key has no setter`**

What am I doing wrong? :cry:

Twiederh, 2024-05-29:
This is how I am using Nodered to have it working:

I would try to leave everything as default, check, if a mosquitto client can read the values from the server the Go-E writes to and try to set the values with Nodered.

Also: this is proven to work, why do you start with sending the values and not getting the values for the grid first and then work on sending them to the Go-E-charger?

[
    {
        "id": "36f44b74645e3f5d",
        "type": "tab",
        "label": "Flow 2",
        "disabled": false,
        "info": "",
        "env": []
    },
    {
        "id": "da25acd5294631b0",
        "type": "mqtt in",
        "z": "36f44b74645e3f5d",
        "name": "",
        "topic": "Powerwall/grid",
        "qos": "2",
        "datatype": "auto-detect",
        "broker": "31dc70c0d498e9bc",
        "nl": false,
        "rap": true,
        "rh": 0,
        "inputs": 0,
        "x": 180,
        "y": 120,
        "wires": [
            [
                "d624a0cff94d5059"
            ]
        ]
    },
    {
        "id": "d624a0cff94d5059",
        "type": "function",
        "z": "36f44b74645e3f5d",
        "name": "PPV an Go-E",
        "func": "var grid = msg.payload;\nvar battery = global.get('battery');\nvar home = global.get('home');\nvar solar = global.get('solar');\nvar backup = global.get('backup');\nvar pwminWatt = global.get('pwminWatt');\nvar pwminSoC = global.get('pwminSoC');\nvar goeMsg = { payload: { pPv:solar, pGrid: grid, pAkku: battery } };\nvar soc = global.get('real_soc' || 0);\n// Pseudoeinspeisung berechnen\nvar pseudogrid = battery + pwminWatt + grid;\n\n// wenn unter Mindestladung Powerwall\nif (soc < pwminSoC) {\n    goeMsg.payload.pAkku = 0;\n    goeMsg.payload.pGrid = pseudogrid;\n}\n// Wenn Mindestladung Powerwall erreicht\nelse {\n    goeMsg.payload.pAkku = battery;\n    // Go-E darf alles ziehen\n    goeMsg.payload.pAkku = 0;\n    goeMsg.payload.pGrid = grid + battery;\n}\n\n// wenn Strom aus der Powerwall gezogen wird\nif (battery > 0) {\n    goeMsg.payload.pAkku = battery;\n}\n\n\nreturn goeMsg;",
        "outputs": 1,
        "timeout": 0,
        "noerr": 0,
        "initialize": "",
        "finalize": "",
        "libs": [],
        "x": 470,
        "y": 120,
        "wires": [
            [
                "f9ea8b774655dc0f"
            ]
        ]
    },
    {
        "id": "f9ea8b774655dc0f",
        "type": "mqtt out",
        "z": "36f44b74645e3f5d",
        "name": "",
        "topic": "go-eCharger1234567/ids/set",
        "qos": "",
        "retain": "",
        "respTopic": "",
        "contentType": "",
        "userProps": "",
        "correl": "",
        "expiry": "",
        "broker": "31dc70c0d498e9bc",
        "x": 940,
        "y": 120,
        "wires": []
    },
    {
        "id": "31dc70c0d498e9bc",
        "type": "mqtt-broker",
        "name": "Mosquitto",
        "broker": "mosquitto.wiederhol.de",
        "port": "1883",
        "clientid": "Nodered",
        "autoConnect": true,
        "usetls": false,
        "protocolVersion": "4",
        "keepalive": "60",
        "cleansession": true,
        "autoUnsubscribe": true,
        "birthTopic": "",
        "birthQos": "0",
        "birthRetain": "false",
        "birthPayload": "",
        "birthMsg": {},
        "closeTopic": "",
        "closeQos": "0",
        "closeRetain": "false",
        "closePayload": "",
        "closeMsg": {},
        "willTopic": "",
        "willQos": "0",
        "willRetain": "false",
        "willPayload": "",
        "willMsg": {},
        "userProps": "",
        "sessionExpiry": ""
    }
]

tennox, 2024-05-30:
Turns out that I was on an old firmware - I did update, but apparently it only updated to an intermediate version and I needed to update again :thinking: 

This works now:
```
mosquitto_pub -h 192.168.178.147 -t go-eCharger/ids/set -m '{"pGrid": 1000, "pPv": 2000, "pAkku":0}'
```

Interesting changelog entry I stumbled upon:
> pgrid target for charging with PV surplus (api key "pgt")
:thinking:

jalmari123, 2025-01-16:
Thank you all! I got It working for pv surplus by using the payload above. 

But does anybody know if it is possible to also use it for load balancing? We only have 3*25A available from the grid, so charging with 3*16A gets sometimes close during the winter with all our other consumption.