goecharger/go-eCharger-API-v2 #207: When providing pGrid and pPv, Go-E does no longer change Amps with latest release of App

Twiederh, 2024-04-18:
Hi,
I have a Tesla Powerwall here, read the values for PV, battery and grid from it and send them to my Go-E charger via MQTT using the go-eCharger/<seriannumber>/ids/set endpoint. Until the latest Go-E App was released yesterday, this was enough for the Go-E  Charger to adopt the charging amps to load with PV only.
With the new App, the charging starts automatically when enough solar power is available, however the ampere value does not change, so if not enough solar power is available, the rest is taken either from the Powerwall or from the grid.

In the app I set the option to charge with solar power (pv surplus "on" in the eco settings) but it looks like the setting "basic" on the main screen instead of "Eco" prevents the Go-E from adopting to available solar power. When I try to switch to Eco, I see in a popup window, that this setting only works with a flexible energy tariff or a Go-E controller.

Since I already have the values, the Go-E controller measures and was able to feed it to the Go-E to have it working with PV surplus, I doe not see the point in buying a Go-E controller. I wonder if this is a bug or it has been done on purpose to make people buy a Go-E controller.

Most people who do not have the technical knowledge anyway will do but it would be nice if for those, who have the technical knowledge, it would be possible to just send the necessary data to the Go-E Charger to do it's thing.

Many regards,
Thomas.

Twiederh, 2024-04-18:
I think I found a workaround: Eco mode can be enabled by checking the flexible tariff option and then manually disabling it in the eco settings.

kalaws, 2024-06-03:
Thank you for the fix. Annoying introduction of feature/bug, unfortunately.

erxbout, 2024-07-04:
I am not sure if I understand you correctly..

If you want to charge pv surplus the mode must be on "ECO" or "Next trip" mode with enabled PV Surplus..
Then it should work if the charger has the grid data available (api-key "ids")

The "Basic" mode is just plug and charge with what current is set in the main menu..

Twiederh, 2024-07-04:
This was a glitch in the first version of the App that has been fixed a while ago, now you can switch on eco mode even when you do not specify an energy tariff.

erxbout, 2024-07-04:
Ok so your issue is fixed? Then I would close it ;)

Twiederh, 2024-07-04:
Yes, the issue is fixed. thank you :-)

closed 2024-07-04 by erxbout