goecharger/go-eCharger-API-v2 #208: Surplusladen funktioniert bei mir nicht ... GoeCharger reagiert nicht

lyncker, 2024-04-21:
Hallo zusammen ich versuche hier das Überschussladen zu aktivieren. Ich habe mir die bestehenden Threads durchgelesen und glaube eigentlich das Prinzip verstande zu haben: 
Über das HTTP V2 Interface schicke ich in der folgenden Reihenfolge Sets an den GOeCharger:
1. http://192.168.x.x/api/set?fup=true
2. http://192.168.x.x/api/set?lmo=4
3. http://192.168.x.x/api/set?ids={"pGrid":-5000,"pPv":1,"pAkku":0} 

Alle Aufrufe werden mit true beantwortet wie der letzte: {"ids":true}.

Den 3. Aufruf wiederhole ich alle 5 Sekunden. 
Leider reagiert der Charger auf nichts. frc steht auf 0 und in der App steht Ladepause. car = 2. 
Wenn ich in der App laden erzwinge, lädt er mit der eingestellten Phase und amp. Aber auch hier reagiert der GoeCharger nicht auf Änderungen von pGrid. So wie ich es verstehe müsste er ja die Ladestärke anpassen und ab einem in der App eingestellten Wert von 4.2 KW auf 3 Phasig umschalten. Er tut aber nichts... 
Woran könnte das noch liegen ?

Twiederh, 2024-05-02:
Wenn pGrid -5000 ist, kann pPv nicht 1 sein. Schicke mal Werte, bei denen die PV-Leistung größer ist als das, was Du ins Netz schickst. Bei mir funktioniert das einwandfrei.

lyncker, 2024-05-02:
ich bin ja nur am testen gwesen... ich habe hier im Forum mehrfach gelesen, dass bei pPv es lediglich drauf ankommt, dass es >0 ist und der eigentliche Wert keine Rolle spielt ? Was soll ich denn übergeben, damit der Charger reagiert?

Twiederh, 2024-05-02:
Bei mir sieht das so aus und es funktioniert:

[image]

Twiederh, 2024-05-02:
Kannst Du denn die Werte, die Du gesetzt hast, so auch wieder auslesen? Ich verwende MQTT und da muss man /ids/set verwenden und nicht /set/ids.

lyncker, 2024-05-02:
ich werde es nochmal mit diesen Werten nachher probieren...vielleicht habe ich pPV einfach nicht hoch genug gesetzt ...

0xFEEDC0DE64, 2024-05-03:
pPv ist nice-to-have für die App visualisierung, pGrid ist das was zählt fürs Regeln, pAkku ist komplizierter (wird von grid abgezogen ausser man möchte die batterie bewusst aufladen oder entladen...)