goecharger/go-eCharger-API-v2 #212: (at least) HTTPv2 and Streaming API: Wallbox unavailable on Wifi loss until next reset

chk2902, 2024-05-02:
At least for me and two of my customers the following problem causes big pain: once the Wifi signal gets lost (here it was an unreliable power line that stopped every other day or so), the wallbox cannot be contacted after Wifi is restored, both connection types fail. The wallbox then needs a reboot (power shutdown or possibly a remote reset via cloud, I did not check) to be available again.

It still works via cloud (my customer tell me).

Either this needs to be fixed (the wallbox becomes responsive again), or we need a heartbeat (once the heartbeat message does not appear any more, the wallbox does a reset, at least of the Wifi part). So the wallbox is not usable for reliable automated charging.

bruma59, 2024-05-03:
or it should go to a predefined state
for example: charging at 6A until predefined SOC reached (daily trip)

chk2902, 2024-05-03:
Hi, it might be it goes to a predefined state - but in the above case that would not help.

But I just checked - now the wallbox Wifi does reconnect here again. I guess the latest firmware update did the job, either "generelle WLAN Verbesserungen" or "Verbindungsproblem mit WPA3 behoben".

closed 2024-05-03 by chk2902