goecharger/go-eCharger-API-v2 #218: How to start charging over API if eCharger is set to "Auth"

eingemaischt, 2024-05-16:
I want the charger to not automatically start charging. So I configured it to closed mode.

My old v2 started charging if I set "allow_charging" / alw to 1.

My new Gemini does...nothing. I have to allow charging via app first. Setting allow_charging to 0 works, though.

mailb84, 2024-05-20:
wie setzt du die Farben der LEDs über die API? Immer wenn ich dies versuche bekomme ich die Fehlermeldung "cch: tried to set api key with invalid json EmptyInput ()"

erxbout, 2024-07-04:
You could achieve the start of charging by changing the mode to basic?
This would be the api-key "lmo" (logic mode) to 3

For debugging you can check modelStatus where u get a reason why charging is allowed/disallowed at that moment

0xFEEDC0DE64, 2024-07-17:
if authentication is set to required, a transaction has to be started to allow the car to charge, use trx=0 for example to start the transaction. The transaction is automatically cleared once the car is disconnected

0xFEEDC0DE64, 2025-09-25:
force state is only considered after the authentication is either disabled, or enabled and authenticated.
When authentication is required but currently not authenticated, the algorithm exits out with not charging and the force state is not considered anymore.
So force state will only help when authentication is off or you already authenticated in some way (rfid card or trx=0)

phreaks, 2025-09-27:
Thx @0xFEEDC0DE64 for your detailed answer. A decision tree is not neccessary :)  ... and it works exactly as you described.