goecharger/go-eCharger-API-v2 #219: Intermittent 502-error from cloudapi calling /api/status

jelgblad, 2024-05-28:
I got a service polling `/api/status` to get the `car`-status and trigger some automations.
When I'm looking at the logs I see that they are filling up with 502-errors from the API-server.

Even tho this isn't causing any real problems (I'll just skip the error response and try again) and the API always seem to get back up again in a few seconds, I'm curious if this might be indicative that a more serious issue with the API-server or even my charger.

Do anyone else get these intermittent errors?

erxbout, 2024-06-26:
Is this happening on a regular basis?
Do you have more information about timings ect?

jelgblad, 2024-07-07:
Here are a few weeks off logs. As you can see it's pretty regular. Timestamps are in UTC+0.

```
Timezone: UTC (0)
2024/06/20 11:06:22 INITIALIZING
2024/06/20 11:06:27 Server responded with code: 404
2024/06/20 11:11:19 Server responded with code: 502
2024/06/20 11:11:24 Server responded with code: 404
2024/06/20 11:11:50 Idle
2024/06/20 11:40:40 Server responded with code: 502
2024/06/20 11:40:45 Idle
2024/06/20 11:43:52 Server responded with code: 502
2024/06/20 11:43:57 Idle
2024/06/20 12:26:24 Server responded with code: 502
2024/06/20 12:26:29 Idle
2024/06/20 12:30:17 WaitCar
2024/06/20 13:19:46 Server responded with code: 502
2024/06/20 13:19:51 WaitCar
2024/06/20 13:33:20 Server responded with code: 502
2024/06/20 13:33:26 WaitCar
2024/06/20 13:49:59 Server responded with code: 502
2024/06/20 13:50:04 WaitCar
2024/06/20 13:52:20 Server responded with code: 502
2024/06/20 13:52:25 WaitCar
2024/06/20 14:15:43 Server responded with code: 502
2024/06/20 14:15:49 WaitCar
2024/06/20 14:27:31 Server responded with code: 502
2024/06/20 14:27:36 WaitCar
2024/06/20 14:33:14 Server responded with code: 502
2024/06/20 14:33:19 WaitCar
2024/06/20 14:46:06 Server responded with code: 502
2024/06/20 14:46:11 WaitCar
2024/06/20 14:59:44 Server responded with code: 502
2024/06/20 14:59:49 WaitCar
2024/06/20 15:08:24 Server responded with code: 502
2024/06/20 15:08:29 WaitCar
2024/06/20 15:37:44 Server responded with code: 502
2024/06/20 15:37:49 WaitCar
2024/06/20 16:04:08 Server responded with code: 502
2024/06/20 16:04:13 WaitCar
2024/06/20 16:27:32 Server responded with code: 502
2024/06/20 16:27:37 WaitCar
2024/06/20 17:17:28 Server responded with code: 502
2024/06/20 17:17:33 WaitCar
2024/06/20 21:00:31 Charging
2024/06/20 21:53:16 Server responded with code: 502
2024/06/20 21:53:21 Charging
2024/06/20 22:24:35 Server responded with code: 502
2024/06/20 22:24:40 Charging
2024/06/20 22:33:44 Complete
2024/06/20 23:43:15 Server responded with code: 502
2024/06/20 23:43:20 Complete
2024/06/21 01:41:24 Server responded with code: 502
2024/06/21 01:41:29 Complete
2024/06/21 01:45:26 Server responded with code: 502
2024/06/21 01:45:31 Complete
2024/06/21 02:53:20 Server responded with code: 502
2024/06/21 02:53:25 Complete
2024/06/21 04:22:40 Server responded with code: 502
2024/06/21 04:22:45 Complete
2024/06/21 04:30:09 Server responded with code: 502
2024/06/21 04:30:14 Complete
2024/06/21 04:38:43 Server responded with code: 404
2024/06/21 04:48:27 Complete
2024/06/21 05:15:45 Server responded with code: 502
2024/06/21 05:15:50 Complete
2024/06/21 08:05:06 Server responded with code: 502
2024/06/21 08:05:11 Complete
2024/06/21 09:08:01 Idle
2024/06/21 09:27:06 WaitCar
[... 822 lines cut ...]
2024/07/07 05:12:57 Complete
2024/07/07 05:29:22 Server responded with code: 502
2024/07/07 05:29:27 Complete
2024/07/07 05:33:14 Server responded with code: 502
2024/07/07 05:33:19 Complete
2024/07/07 07:00:20 Server responded with code: 502
2024/07/07 07:00:25 Complete
2024/07/07 07:35:10 Server responded with code: 502
2024/07/07 07:35:15 Complete
2024/07/07 08:34:39 Server responded with code: 502
2024/07/07 08:34:44 Complete
2024/07/07 08:36:10 Server responded with code: 502
2024/07/07 08:36:15 Complete
2024/07/07 08:43:36 Server responded with code: 502
2024/07/07 08:43:41 Complete
2024/07/07 08:58:16 Server responded with code: 403
2024/07/07 08:58:37 Complete
2024/07/07 09:24:34 Server responded with code: 502
2024/07/07 09:24:39 Complete
2024/07/07 10:00:05 Charging
2024/07/07 10:03:12 Server responded with code: 502
2024/07/07 10:03:17 Charging
2024/07/07 12:19:22 Server responded with code: 502
2024/07/07 12:19:27 Charging
```

erxbout, 2024-07-08:
Interesting..

Do I understand correctly that your polling interval is 5 seconds?
Will do some testing there

0xFEEDC0DE64, 2024-07-08:
what do the response headers and the response body say?