goecharger/go-eCharger-API-v2 #221: modbus: Read Holding Registers does not behave as expected

73jn, 2024-06-04:
Hello,
I have a GO-E charger with the latest version 56.1.
I need to retrieve information from the charger on its Modbus server. Using a Modbus TCP client, I just noticed that the behavior of a single register and several registers is not the same only on some registers.

For example, reading register 206 (Led Brightness) gives me the value 255. However, if I read registers 205 and 206, register 206 (Led Brightness) equals -1, whereas it should be 255.

Read Holding Register 206 :
[image]

Read Holding Registers 205 & 206
[image]

Read Holding Registers 206 & 207
[image]

/jean

erxbout, 2024-07-08:
I am sorry I can not replicate your issue..
When I read the registers I get this output

[image]

Maybe the output is different if you test it with a different tool?

73jn, 2024-07-08:
The error seems to occur when you read registers 205 and 206, not 206 and 207.

erxbout, 2024-07-08:
Ah I overlooked that part..
Now I can reproduce it.. seems like this is always the case when reading more than one register before 206 (for example 200 - 210)

We will note it as a bug but I can not promise you a fast resolve..

erxbout, 2024-09-13:
The issue might be fixed now with the new beta release 56.9
Had no time to test it yet..