goecharger/go-eCharger-API-v2 #225: Überschussladen für Dumme

SofaHomie, 2024-06-16:
Hallo zusammen.
ich bin ziemlich unbewandert in allem, was mit programmierungen einher geht. daher habe ich 2 fragen für blöde (sorry).
ich möchte überschussladen aktivieren, weiß aber nicht, wo ich den vorgegebenen code hinterlegen muss um pgrid weiterzugeben. kommt der code in die "service.yaml" unter "config/goecharger_api2"? oder wo muss ich den einfügen?

als zweite frage: mein wechselrichter ist von sungrow. er gibt mir nicht einen einzelnen +/- wert für das laden in das oder aus dem netz, sondern hat 2 separate werte. beide natürlich positiv. sehe ich es richtig, dass ich hier nur den sensor hinterlege, der für das einspeisen ins netz zuständig ist und dann über den "float"-befehl den wert anpasse?

danke für die hilfe und sorry, dass ich keinen plan habe^^

0xFEEDC0DE64, 2024-06-24:
Um was für eine software gehts überhaupt? Wir bieten keine service.yaml an, soweit ich weiß?

Falls software bastellösungen nicht gewünscht sind, reicht nicht einfach ein go-e controller mit messklemmen am einspeisepunkt aus?

marq24, 2024-06-24:
Das ist eine Home Assistant Automation [dafür muss man zunächst eine eigenen Server betreiben https://www.home-assistant.io/ ] und dann kann man (mit ein wenig Bereitschaft zum Lesen und Einarbeitung) das ganze ans Fliegen bringen - wenn Du aber (wie Du selbst schreibst "absolut keine Ahnung" hast, ist der Vorschlag von @0xFEEDC0DE64 wohl der bessere Weg für Dich.

SofaHomie, 2024-06-25:
ah, das hab ich wohl etwas undeutlich ausgedrückt, sorry. der home assistant läuft. sungrow (wechselrichter) hab ich als integration drin (und bekomme auch die sensor-werte). die api_v2-integration habe ich auch drin. auch das funktioniert, auch da bekomme ich die sensorwerte. ich kann auch die paar zeilen code verstehen und ich kann auch den korrekten sungrow-sensor hinterlagen, um den pgrid-wert weiterzugeben und pv-überschuss zu nutzen. das einzige, was ich nicht verstehe, an welcher stelle des home.assistant ich den code hinterlegen muss. es gibt ja mehr als eine möglichkeit. ich würde das "studio code server" plugin nutzen, um den code einzugeben. aber muss das in einer der *.py-dateien oder *.yaml-dateien unter "custom_components -> goecharger_api2" oder muss das in die configuration.yaml? diese hilfe bräuchte ich, da dass, was ich bisher versucht habe, nicht funktioniert hat.

tut mir leid, wenn ich lästig bin. aber für jede info bin ich dankbar.

schinagl, 2024-06-25:
Ich glaub das wäre eher eine Frage im Home Assist Forum?

marq24, 2024-06-25:
... @SofaHomie Du bist hier im go-eCharger GitHub Projekt - derzeit gibt es von go-e keine Software für HA oder einer andere HomeAutomation Platform... Wenn Du Unterstützung für eine HomeAssistant Integration benötigst, dann solltest Du dort bitten, wo Du die Integration herhast - aus gut unterrichteten Quellen "weiss ich zufällig" das dort ein sehr deutlicher Hinweis zu finden ist, das die Integration in keinerlei Verbindung mit go-e steht... lesen - verstehen - handeln...

erxbout, 2024-06-26:
Der go-e Controller ist genau für diesen Zweck gemacht und lässt sich entsprechend mit dem Charger verbinden.
Es wird hierbei keine Programmierkenntnis benötigt und das Überschussladen lässt sich so auch gut mit einem dynamischen Bezugstarif kombinieren (man bedenke Winter)

Ich würde auch persönlich davon abraten Code irgendwo hinzukopieren wenn nicht bekannt ist was das genau tut und wo man das hinkopieren muss..

closed 2024-06-26 by erxbout

SofaHomie, 2024-06-26:
verstanden. besten dank für die zeit.