goecharger/go-eCharger-API-v2 #226: Überschuss Laden - die nächste

wrzlbrmft1, 2024-06-25:
Hallo,
ich bin am Verzweifeln, denn der goE-Charger lädt mit geringer Leistung weiter, obwohl kein PV-Überschuss erzeugt wird.

Typ: go-eCharger_V4 
Firmware: 56.1

Meine Werte:
-  pGrid: 1377
-  pPv: 402
-  pAkku: 4
-  fup: true
-  lmo: 4
-  psm: 0
-  frc: 0
-  modelStatus: 20

In der APP steht folgendes: "Laden mit 1,35 kW, weil Ladepause nicht erlaubt ist".

Normalerweise müsste doch bei Netzbezug das PV-Überschuss-Laden aufhören?
Welche Einstellung verursacht die Meldung, dass Ladepause nicht erlaubt ist?

ich finds echt schade, dass die Grid-Schnittstelle so rudimentär dokumentiert ist.

Gruß in die Community . . .

erxbout, 2024-07-03:
Ist der api-key "acp" gesetzt?

Auszug aus der doku:

| Key        | R/W        | Type                         | Category      | Description                                                                         |
| ---------- | ---------- | ---------------------------- | ------------- | ----------------------------------------------------------------------------------- |
| acp        | R/W        | bool                         | Config        | allowChargePause (car compatiblity)                                                 |

wrzlbrmft1, 2024-07-05:
> Ist der api-key "acp" gesetzt?
> 
> Auszug aus der doku:
> Key 	R/W 	Type 	Category 	Description
> acp 	R/W 	bool 	Config 	allowChargePause (car compatiblity)

Danke - an dem Schlüssel hats gehangen. 

Nun funktioniert das PV-Laden

erxbout, 2024-07-08:
Sehr gut, bitte gerne :+1:

closed 2024-07-08 by erxbout