goecharger/go-eCharger-API-v2 #227: Sign Parameter for PV Surplus

charlyc79, 2024-07-01:
Hello everyone,
I have a question for the sign of the parameter psh, sh, pgt, fst, spl3 and zfo. 
The signal pgrid for the grid power has to be negative, if my pv produces more power than is needed. 
The signal deltap is positive. I think the calculation is deltap = - ( pgrid - pgt)

But when does the session start (parameter fst): pgrid- pgt < fsh or deltap < fsh? First fsh has to be negative for PV Surplus, Second one has to be positive.
And the same questions fpr psh, sh, spl3 and zfo. 

Are there some mathematical formulas to understand better?

Many thanks.

Overall the integration is great for my Cupra Born and my GO-E Charger V4 and  SW 56.1.

erxbout, 2024-07-04:
I am not sure what you are asking here.. Is it clarification about positive or negative numbers?

As far as I know we use pGrid for pv surplus and pAkku only comes into play if you have a battery and adjust some more settings..

When pGrid is negative it is handled as export into grid
When pGrid is positive it is handled as import from grid

erxbout, 2024-09-25:
@charlyc79 do you need additional information on this issue?