goecharger/go-eCharger-API-v2 #229: Go-e Firmware 56.8 and API-v2 / Modbus

rainer4, 2024-07-07:
### Discussed in https://github.com/goecharger/go-eCharger-API-v2/discussions/228

<div type='discussions-op-text'>

<sup>Originally posted by **rainer4** July  4, 2024</sup>
After installing Go-e Firmware 56.8 I have some Modbus issues:
e.g. register 100 (car state) was 1 "idle" in 56.2 when no car was connected, with 56.8 i receive 0 "unknown/defect"
other registers like 299 (Ampere_Volatile) worked
Are there any changes regarding Modbus registers with 56.8 Firmware?</div>

erxbout, 2024-07-08:
It should still work normally..
What is the state of the error register? (107)

rainer4, 2024-07-08:
I had to go back to Firmware 56.2 but as I remember error register 107 was 0, but car state was 0 "unknown/defect" (no car was connected and no cable was connected)

erxbout, 2024-07-08:
Could you provide the serial number and the timeframe where this happened?

rainer4, 2024-07-08:
I just installed update 56.8 again to reproduce the error, but this time everything works fine. Last time I could reproduce the error twice. Sorry, please close this issue.

erxbout, 2024-07-08:
No worries, glad that everything works now ;)

closed 2024-07-08 by erxbout