goecharger/go-eCharger-API-v2 #233: Go-e Firmware 56.8 and API-v2 / PV surplus charging 5A current cap per phase

karliwalti, 2024-07-15:
After I updated 56.8, I observed that the PV optimized charging  not longer worked properly. 
In my setup (huawei PV, homeassistant with API2, go-e), the charger caps the current on each phase to max 5A.  When there is enough power from PV (4.2kW),  the phase switch to 3 phases works, but still only 5A on each phase. Downgrade to 56.1 did make it work again, but unfortunately the sun was then soon gone to do further tests.

Here some more details to illustrate the issue: 

You can nicely see how PV did not go into car:
[image]
Go-e did see the surplus (so API transmits correct values), but charger did nothing:
[image]

 L1 was limited to 5A (I did downgrade of the FW  around 14:30 )
[image]

Settings were correct:
[image]
[image]
[image]

There have been no reported errors. 

It does not seem to be linked to the API2 integration in HA (this is a copy of what I originally posted there https://github.com/marq24/ha-goecharger-api2/issues/11#issuecomment-2227462588) but more likely to some firmware changes that mess with the PV surplus charging algorithm. 
Would be great to have access to the FW changelog. The app does not show a changelog for 56.2 and 56.8. 

The weirdest thing is that the API can only limit current to minimum of 6A `number.goe_259493_ama`  (and mine was set to 16A), so not sure how a cap of 5A is even possible.

peterpoetzi, 2024-07-15:
Hi,

you can have a look at the grafana integration on the data export website. To access it, go to the mobile app in "Information" and click the "Download information" link on the bottom. You will get a website where you can view past charging sessions. For every charging session, you can view some graphs of the allowed current, and the actual current per phase.

The charger itself cannot signal 5A to the car. I had a look at the support data and it shows, the car was constantly given a 16A signal in that timerange. So the limit of 5A comes from the car and not from the charger.

Peter

karliwalti, 2024-07-15:
thank you for the info, did not know about this little grafana feature.

I would not have an explanation for why the car would not allow higher currents...and why after the downgrade it didnt anymore.

mr-gosh, 2024-07-17:
> you can have a look at the grafana integration on the data export website. To access it, go to the mobile app in "Information" and click the "Download information" link on the bottom. You will get a [...]

What if the collection stopped after the 8th recording back in 2022? Does this feature only work in combination with some other setting?

erxbout, 2024-07-23:
Any kind of new information? @karliwalti

karliwalti, 2024-07-24:
I am sorry, I do not currently have a car at hand to test it.

karliwalti, 2024-08-27:
I finally got to test the configuration again. First, old firmware all worked as expected (25.8.). Updated firmware yesterday. Will keep you posted when Sun reappears ...

karliwalti, 2024-08-28:
So, finally, a sunny day....
I confirm that firmware  56.8 works again and sending API2 calls to track surplus solar is responsive. I no longer can reproduce above pattern.  The 5 A per phase will remain a mystery.
Phase switching also works. 

Still, the following behaviour is illogical and might be worth another ticket. 

The requested current limit `amp` is overriding the surplus dynamics. This means, if `amp` is set to 6A, the charger would charge only at 1.6kW (6A on 1phase) until there is enough surplus to charge at 4.2kW (6A on 3 phases). In the surplus calculation, only the max current limit `ama` should be used?

erxbout, 2024-09-25:
We treat the amp setting as the customer wish for charging..
While charging dynamically we factor in this wish as we limit the current to the minimum value of all the available current limits..
I do not think that we should change that or we will potentially break other customers wishes..

Please just increase the amp setting according to your wishes