goecharger/go-eCharger-API-v2 #241: Homeassistant MQTT autodiscover (missing entities)

mmick08, 2024-09-06:
Hi Guys,

First of all, thank you for integrating Homeassistant MQTT directly in your firmware. A great step underlining your dedication to open source.

I just wanted to point out the following topics:

Missing entities:

Even if the auto discover works very well several important entities to do home automation are missing.
i.e. "Car Connected" (is the cable plugged in or not)
"Configured phases" (how many phases are currently used)
"phase switching" (possibility to choose between automatic, 1-phase, 3-phases)

Wrong entity details:

As for now the "total_energy_charged" cannot be used within the Energy dashboard as state_class: total_increasing is missing for the entity.

BR Manuel

CommanderRedYT, 2024-09-16:
Hi Manuel, I took a look at your issue.

- For the "Car Connected", there already exists the "Car State" in Homeassistant, so that should be covered. It should be 1:1 the Api-Key `car`.
- For the phases, I will expose the Api-Key `pha`.
- Phase switching will also be implemented.
- Also, I added `total_increasing` to `total_energy_charged`.

I do not know yet when we release this, but I will write here again!

Thanks for your feedback :)

mmick08, 2024-09-16:
Hi,

Unfortunately, the 'car' variable is not properly exposed. The only available boolean is 'allowed to charge,' but its value is sometimes 'True' even when the car is not connected.

[image]

Thanks for the rest. 👌🏼

mmick08, 2024-09-17:
Hi! 

why posting the same message again ? :) 

"Car" or Car State is NOT available. See Screenshot. Thats what i said.

Br Manuel

CommanderRedYT, 2024-09-17:
You are only checking the sensors, not the config.

CommanderRedYT, 2024-09-17:
Hm, my instance for some reason showed it despite the error in #244. After this is fixed, it should be visible.

mmick08, 2024-10-07:
Thank you for the latest Beta. Now as you said "Car state" is available BUT its not the connected state at all! 🙈 

Connected State should be a boolean either True or False not something like "Charging, Idle, WaitCar..."

[image]

erxbout, 2024-12-17:
We do not have an api key that maps the car_state to a booleon car_connected true/false..
And I am not sure if it makes sense to implement one as the enum we have for car_state carries more information than true/false

You could use a helper to generate a true/false value
It would be true if Charging, WaitCar or Complete..

mmick08, 2024-12-17:
already did that in the meantime.
thanks

lukastribus, 2025-10-20:
I suggest mapping the API Key `amp` in HA auto discovery. This is the equivalent of the charging speed selector in the Go-E Charger App (as opposed to the `ama` key that is already mapped in HA autodiscovery but used for other things).

Here's what works for me in HA:

```
homeassistant/number/go-e_serial/set_charging_current/config
{
  "name": "Set Charging Current",
  "~": "go-eCharger/serial/amp",
  "stat_t": "~",
  "cmd_t": "~/set",
  "unit_of_meas": "A",
  "ent_cat": "config",
  "min": 6,
  "max": 32,
  "uniq_id": "go-e_serial_set_charging_current",
  "dev": {
    "mf": "go-e",
    "mdl": "go-eCharger_V4",
    "name": "go-eCharger_serial",
    "ids": [
      "go-e_serial"
    ],
    "sw": "59.4"
  }
}
```