goecharger/go-eCharger-API-v2 #242: Request maximum charging power using OCPP

mincequi, 2024-09-07:
Hi there,

how can i request the maximum charging power using OCPP? Is this supported?

Thanks a lot.

0xFEEDC0DE64, 2024-09-09:
we only support limiting the current, so the power depends on if the charger has voltage supplied on all three phases or only one, and if the car can draw current on 1 or 2 or 3 phases, we might allow limiting the actual power over OCPP Smart Charging profiles in the far future by using our pv surplus control loop, but that is not wired into ocpp yet and is also not planned yet.

mincequi, 2024-09-10:
Sorry for not being clear enough.
I mean, if there is a way to request the maximum power/current the wallbox actually can deliver?
11kw vs 22kw / 16A vs 32A.
Is there a way to request this via OCPP?

also: can one request voltage using OCPP? There seems to be an appropriate measurand key defined in OCPP 1.6J for MeterValues: _Voltage_

0xFEEDC0DE64, 2024-09-10:
I have not yet exposed the device variant on ocpp yet, should it be a custom readonly configuration key for GetConfiguration or do you have a better way?

setting the measurands for voltage too should work

mincequi, 2024-09-14:
Reading voltages works, thanks!

Having a custom readonly key for GetConfiguration sounds reasonable to me.
Alternatively, this info is already provided via BootNotification (as chargePointMaxPower or chargePointMaxCurrent).