goecharger/go-eCharger-API-v2 #243: RemoteStartTransaction gets rejected.

mincequi, 2024-09-08:
Hi there,

i have a go-eCharger V4 with this Firmware 56.9.
However, the following RemoteStartTransaction gets rejected:

[2,"RemoteStartTransaction 31","RemoteStartTransaction",{"chargingProfile":{"chargingProfileId":1,"chargingProfileKind":"Absolute","chargingProfilePurpose":"TxProfile","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"limit":6,"numberPhases":1,"startPeriod":0}]},"stackLevel":0},"idTag":"user"}]

Do you have any idea, what could be wrong here?

Thanks a lot.

0xFEEDC0DE64, 2024-09-09:
I dont think this is valid OCPP at all, why mixing RemoteStartTransaction with a charging profile? Does the spec allow this?

mincequi, 2024-09-10:
According to this spec: https://groups.oasis-open.org/higherlogic/ws/public/download/58944/ocpp-1.6.pdf/latest

**Yes**, that should work.

or: which way would you suggest to continuously adapt the allowed charging power/amps for PV surplus charging?

0xFEEDC0DE64, 2024-09-10:
gotcha! I havent seen RemoteStartTransaction mixed with a chargingProfile yet, we currently only support SetChargingProfile and ClearChargingProfile.

I will put your example on my todo list

EnvironCMO, 2024-10-14:
Does anyone have a working example of a SetChargingProfile command? My commands are accepted but unfortunately the wallbox does not execute these commands