goecharger/go-eCharger-API-v2 #244: Home Assistant 2024.9 - MQTT - fehlende DeviceClass „enum“

maxxnet2, 2024-09-11:
Hallo,
seit 2024.9 erhalte ich in Home Assistant den MQTT-Fehler, dass bei jenen Sensoren, welche die Option „options“ verwenden, die DeviceClass nicht definiert ist. Erwartet wird hierbei die DeviceClass „enum“. Den selben Fehler hatte ich auch bereits bei einem anderen Add-on. Dort hat der Entwickler daraufhin eingegriffen und die DeviceClass ergänzt. 

Logger: homeassistant.components.mqtt.mixins
Quelle: components/mqtt/mixins.py:166
Integration: MQTT (Dokumentation, Probleme)
Erstmals aufgetreten: 12:50:59 (2 Vorkommnisse)
Zuletzt protokolliert: 12:50:59

Error 'The option `options` can only be used together with device class `enum`, got `device_class` 'None'' when processing MQTT discovery message topic: 'homeassistant/sensor/go-e_263090/car_state/config', message: '{'unique_id': 'go-e_263090_car_state', 'options': ['Unknown', 'Idle', 'Charging', 'WaitCar', 'Complete', 'Error'], 'device': {'manufacturer': 'go-e', 'identifiers': ['go-e_263090'], 'sw_version': '56.8', 'name': 'go-eCharger_263090', 'model': 'go-eCharger_V4'}, 'command_template': "{% set o = {'Unknown': 0,'Idle': 1,'Charging': 2,'WaitCar': 3,'Complete': 4,'Error': 5} %}{{ o[value] if value in o else value }}", 'state_topic': 'go-eCharger/263090/car', 'name': 'Car State', 'value_template': "{{ ['Unknown','Idle','Charging','WaitCar','Complete','Error'][value_json|int ]}}", 'platform': 'mqtt'}'
Error 'The option `options` can only be used together with device class `enum`, got `device_class` 'None'' when processing MQTT discovery message topic: 'homeassistant/sensor/go-e_263090/error_state/config', message: '{'unique_id': 'go-e_263090_error_state', 'options': ['None', 'FiAc', 'FiDc', 'Phase', 'Overvolt', 'Overamp', 'Diode', 'PpInvalid', 'GndInvalid', 'ContactorStuck', 'ContactorMiss', 'StatusLockStuckOpen', 'StatusLockStuckLocked', 'FiUnknown', 'Unknown', 'Overtemp', 'NoComm', 'CpInvalid'], 'device': {'manufacturer': 'go-e', 'identifiers': ['go-e_263090'], 'sw_version': '56.8', 'name': 'go-eCharger_263090', 'model': 'go-eCharger_V4'}, 'command_template': "{% set o = {'None': 0,'FiAc': 1,'FiDc': 2,'Phase': 3,'Overvolt': 4,'Overamp': 5,'Diode': 6,'PpInvalid': 7,'GndInvalid': 8,'ContactorStuck': 9,'ContactorMiss': 10,'StatusLockStuckOpen': 11,'StatusLockStuckLocked': 12,'FiUnknown': 13,'Unknown': 14,'Overtemp': 15,'NoComm': 16,'CpInvalid': 17} %}{{ o[value] if value in o else value }}", 'state_topic': 'go-eCharger/263090/err', 'name': 'Error State', 'value_template': "{{ ['None','FiAc','FiDc','Phase','Overvolt','Overamp','Diode','PpInvalid','GndInvalid','ContactorStuck','ContactorMiss','StatusLockStuckOpen','StatusLockStuckLocked','FiUnknown','Unknown','Overtemp','NoComm','CpInvalid'][value_json|int ]}}", 'platform': 'mqtt'}'

sffbigmac, 2024-09-12:
exact same issue here as well, moved to 2024.9 and because of the above issue the two corresponding sensors become unavailable.

CommanderRedYT, 2024-09-17:
Hi, this will be fixed in one of the next releases!