goecharger/go-eCharger-API-v2 #246: V 56.8 MQTT sends Homeassistant although HA is switched off

juergen852, 2024-09-16:
Firmware 56.8 sends homeassistant topic + data altough "MQTT homeassistant aktiviert" is switched off in the app .

Checked on my 10 chargers...

CommanderRedYT, 2024-09-17:
Was it turned on at any given time? If so, you will have to manually delete the topics from your mqtt server because the retain flag is set.

erxbout, 2024-09-25:
This is the same problem as in #231

juergen852, 2024-09-25:
@CommanderRedYT 
It might have been turned on on 1 or 2 chargers for testing, but not on all.
I deleted all messages on my server.
All chargers re-appear with homematic, although not activated in e-go app.

[image]

0xFEEDC0DE64, 2024-10-18:
It was fixed 2 or 3 beta versions in the past already

closed 2024-10-18 by 0xFEEDC0DE64