goecharger/go-eCharger-API-v2 #247: V56.8 and V56.9 beta - no access to internet via Wifi for LTE chargers

sseigneurin, 2024-09-27:
We have 3 LTE chargers which are able to connect to a Wifi access point and get the IP address but they are not able to communicate to the internet via Wifi. As soon as we switch off the LTE connection, the Cloud connection stops as well as the OCPP connectiion.

Is it an expected behavior and is there a workaround?

sseigneurin, 2024-09-27:
Note: the wifi works well for a PC or Mac connecting to the internet (DHCP is used).

erxbout, 2024-12-17:
Is this still happening with 57.1?

sseigneurin, 2024-12-17:
We have identified that it's linked to a parameter on the 4G router that we use, "Software flow offloading". I don't know what it exactly implies but if we leave this parameter ON, the charger doesn't manage to reach the Internet. Turning the parameter OFF enables the charger to reach the internet. I can't tell if it is still in version 57.1.