goecharger/go-eCharger-API-v2 #248: Auto wird "abgehängt" gemeldet obwohl es an der WB angeschlossen ist - Firmware 56.8

chk2902, 2024-10-01:
Situation: Auto ist angehängt, aber voll ("car":4) - weiß nicht, ob das einen Einfluss hat - dann initiiere ich eine Phasenlimitierung, das Auto wird dann als abgehängt gemeldet ("car": 1), 10 Sekunden später wieder als verbunden ("car":3). 

... was ich schade finde, denn beim Abhängen werden in meiner App diverse Parameter auf Voreinstellung zurückgesetzt. D.h. wenn ich die von der Voreinstellung ändere und die Ladung läuft, dann eine Phasenlimitierung ein- oder ausgeschaltet wird, wird alles wieder auf die Voreinstellung zurückgesetzt.

Muss irgendwie neu sein, das ist bisher m.W. nicht passiert - wobei ich normalerweise auch die Parameter nicht ändere.

Da mittlerweile der Tesla auch mindestens ein Update hinter sich hat, kann ich nicht garantieren, dass der nicht auch noch beteiligt ist.

chk2902, 2024-10-01:
Das passiert anscheinend aber nicht immer - vorhin hat er die Phasenbegrenzung eingeschaltet, später ausgeschaltet - und der "car"-Zustand ging nicht auf 1 zurück. Seltsam.

erxbout, 2024-12-18:
Also bei manueller Phasenbegrenzung ändert sich der car state auf Idle (1)?
Da hardwareseitig ein "ausstecken" simuliert wird würde das Sinn machen (10 sekunden Abstand, damit das Auto damit klar kommt was hier gerade passiert)

Was bei automatischer Phasenumschaltung anders ist müsste getestet werden..

Ganz andere Frage: Wenn das Auto voll ist und dann Phasenwechsel durchführen will, was hätte das für einen use case? Vorheizung über "Landstrom" würde mir jetzt spontan einfallen.. ^^

pschlein, 2025-01-22:
Ein ähnliches Verhalten ist auch beim Starten der Box festzustellen. 
Ich habe meine Boxen mittels Schütz vom Netz getrennt um die Standby Verbräuche zu reduzieren.
Gelegentlich kommt es nach dem Hochfahren (Strom einschalten) vor, dass Car=1 retourniert wird, obwohl das Auto angesteckt ist. Der Zustand kann aber nicht auf Wunsch Reproduziert oder Erzwungen werden.

Ein Neustart der Box mittels /api/set?rst=1 löst zwar das Problem, kann aber nicht automatisiert werden da es potentiell eine Endlos-schleife wäre, wenn tatsächlich kein Auto dran hängt.

Neben dem Car=1 kommt hier und da auch (zusätzlich) ein modelStatus=24, was ebenfalls durch ein Reboot gelöst werden kann.