goecharger/go-eCharger-API-v2 #249: SetChargingProfile did not work

EnvironCMO, 2024-10-07:
Hello,

I send this to my go-e Charger:

`[
  2,
  "316414082",
  "SetChargingProfile",
  {
    "connectorId": 1,
    "idTag": "no-card",
    "csChargingProfiles": {
      "chargingProfileId": 1,
      "transactionId": 0,
      "stackLevel": 0,
      "chargingProfilePurpose": "TxProfile",
      "chargingProfileKind": "Absolute",
      "chargingSchedule": {
        "duration": 82800,
        "startSchedule": "2024-10-07T17:03:02.379399+02:00",
        "chargingRateUnit": "A",
        "chargingSchedulePeriod": [
          {
            "startPeriod": 0,
            "limit": 0
          },
          {
            "startPeriod": 32217,
            "limit": 16
          },
          {
            "startPeriod": 43017,
            "limit": 0
          }
        ]
      }
    }
  }
]`

And get this back:
`[
  3,
  "316414082",
  {
    "status": "Accepted"
  }
]`

But the Pofil is not used. When I plug in the vehicle, charging starts immediately. Have I done something wrong, do I have to make special settings or is it a bug?

EnvironCMO, 2024-10-09:
I use a go-e Charger Gemini flex 11 kW
With Software Version 56.8

DavidKarlas, 2024-11-11:
Could you try with something like this?
```
[
    2,
    "448517309",
    "SetChargingProfile",
    {
        "connectorId": 1,
        "csChargingProfiles": {
            "chargingProfileId": 20,
            "chargingProfileKind": "Absolute",
            "chargingProfilePurpose": "TxDefaultProfile",
            "chargingSchedule": {
                "chargingRateUnit": "A",
                "chargingSchedulePeriod": [
                    {
                        "limit": 7,
                        "numberPhases": 3,
                        "startPeriod": 0
                    }
                ],
                "startSchedule": "2024-11-11T03:30:31Z"
            },
            "stackLevel": 20
        }
    }
]
```

DavidKarlas, 2024-11-11:
I also noticed, that this will not update mobile app max amps, and mobile app max amps will always be respected, so for example if I have mobile app max amps set to 10A, I can set via OCPP amps between 6A and 16A, but car will only get 10A, never 12A, but if OCPP sets 7A, car will get 7A, up to 10A, but not past that. So when testing this, make sure to set mobile app amps to 16A