goecharger/go-eCharger-API-v2 #250: PV Überschussladen funktioniert nicht richtig bei automatischer Phasenumschaltung

mpichlba, 2024-11-06:
Ich schicke im 5 Sekunden Takt mit einem Script die Werte vom Fronius Wechselrichter zur GoE API. (inkl batterie infos)
zb:
Grid Power (P_Grid): -13 W
Load Power (P_Load): -2950 W
Battery Power (P_Akku): -1244 W
Solar Power (P_PV): 4297 W
http://goe/api/set?ids=%7B%22pGrid%22%3A-13%2C%22pPv%22%3A4297%2C%22pAkku%22%3A-1244%7D

Das PV Überschussladen startet und tut auch was es soll. 
Was jedoch manchmal passiert ist, dass die PV zu wenig Strom liefert für 3 Phasen und dann zusätzlich die Hausbatterie angezapft wird um auf die erforderlichen Minimumwerte für 3phasiges Laden zu kommen anstatt auf 1-Phasig zu wechseln.
warum ist das so?
der einzige Workarround ist, dass ich die Wallbox einstelle dass nur 1-phasig geladen wird.

schinagl, 2024-11-06:
Ich erinnere mich dass @0xFEEDC0DE64 mal das Wort 'wolkensmoothing' 😂 verwendet hat und ich denke du wartest nur nicht lang genug dass auf 1 phasig geschaltet wird. 
Moral: da gibts eine grace period wo nicht umgeschaltet wird. 

Gleiches hab ich beobachtet wenn du unter 1.4kw PV fällst. Dann wird auch eine Zeit aus Akku und Netz gefahren bevor ganz abgeschaltet wird.

mpichlba, 2024-11-06:
Danke für das rasche Feedback @schinagl. Macht ja auch Sinn dass man nicht die ganze Zeit hin- und herschaltet. kann man diese Grace Period wo einsehen?

schinagl, 2024-11-06:
Ich vermute das ist Teil des Regelalgorithmus, und das weis   @0xFEEDC0DE64. 

Gefühlt ist das aber 1 oder 2 Minuten. 

Wie lang ist es wirklich?

closed 2024-11-07 by mpichlba