goecharger/go-eCharger-API-v2 #253: autodiscovery of MQTT entity "Set Next-Trip Energy Limit" doesn't set max value

sr258, 2024-11-29:
Hi,

I've enabled MQTT in my go-e Charger, connected it to Mosquito running as a plugin in Home Assistant and enabled the MQTT integration of Home Assistant.

Home Assistant automatically discovers the go-e Charger entities, but the entity that allows you to set the next trip energy limit ("Set Next-Trip Energy Limit", go-e_XXXXXX_set_nexttrip_energy_limit), is unusable, as there's an implicit maximum value of 100.0 in Home Assistant for Number entities. As the energy limit is set in Wh, you need to set it to something much bigger like 20000, which isn't possible because Home Assistant's validation rejects the value.

It would be great if the auto-discovery data of the entity contained an explicit sensible maximum value like 150000.

erxbout, 2024-12-17:
this should be fixed in version 57.1, please update and close this issue if it works correctly ;)

sr258, 2024-12-17:
Thanks for the quick fix! I’ve already noticed the change in the update and it works as expected.

closed 2024-12-17 by sr258

erxbout, 2024-12-17:
Nice to hear thanks! ;)