goecharger/go-eCharger-API-v2 #254: HTTP-API Änderung zwischen Firmware 56.8 und 57.1?

pd82wi, 2024-12-15:
Ich verwende Loxone zur Steuerung des go-e Chargers. Über die Register ALLOW, AMPERE_VOLATILE und PHASE_SWITCH_MODE kann ich mit der Firmware 56.8 (und auch bei älteren Versionen) die Phasenumschaltung und die Stromstärke steuern. Nach dem Update auf 57.1 funktioniert die Steuerung nicht mehr. Der go-e Charger reagiert nicht mehr auf die Kommandos. Hat es hier eine API Änderung gegeben, die noch nicht in der Dokumentation vorhanden ist? Info: nachdem ich wieder zurück auf 56.8 gegangen bin, hat alles wieder einwandfrei funktioniert.

pd82wi, 2024-12-17:
Ja, es geht um die Modbus Integration von: https://github.com/goecharger/go-eCharger-API-v2/blob/main/modbus-de.md
Danke!

erxbout, 2024-12-18:
Wenn ich die seriennummer habe könnte ich ein update aufspielen welches einen möglichen Fix für dieses issue hätte..

So oder so landet dieser Fix im nächsten Release, für das gibts aber noch kein Datum

pd82wi, 2024-12-18:
Danke für das Angebot, ich bleibe aber erstmal auf 56.8 da es damit bisher problemlos funktioniert. Ich aktualisiere dann auf die neue Version sobald sie verfügbar ist.

MichaelFrech, 2024-12-27:
Auch bei mir selbe Situation :
Integration mit Loxone mit Firmware 57.1 keine ModbusTCP Kommunikation ( kein beschreiben der Register möglich) downgrade auf 56.8 und Funktion wieder ok , meine auch das bei 56.9 kein Problem bestand kann ich aber nicht zu 100% sicher sagen.

Danke und schöne Grüße.

tmcrash, 2024-12-27:
Ich hatte über die Feiertage ein wenig getestet und kann bestätigen, dass das Problem schon in 57.0 eingezogen war.
Scheinbar wird nun auch bei nicht angeschlossenem Fahrzeug der Modbus Register 200_Allow auf 1 gesetzt. Da müsste ich dann auch meine Visualisierung anpassen (Knopf für Adhoc Laden springt dann auf "EIN" um).
Wäre es möglich, eine Beta mit dem Fix zu testen, oder eine individuelle FW (Hardware V3 mit S/N 102253)?
Zurzeit kann man nämlich nur FW 56.8 benutzen und die hat das bekannte Problem mit den Registern 204_Cable_Lock_Mode und 206_LED_Brightness. Alles danach hat schwerwiegendere Probleme (WLAN Leistung oder nun Register 200 nicht mehr beschreibbar).

gpitzer, 2025-01-04:
Ich habe das gleiche Probelm, denke aber dass es mit dieser FW schon funktioniert hat.
Leider kann ich nicht mehr auf die 56.8 downgraden, keine Ahnung warum. Ich komme nur auf 56.9. Beim Downgradeversuch von 56.9 auf 56.8 macht er ein Upgrade auf 57.1 Für eine Beta Version wäre ich auch sehr froh. Meine ID vom v4 charger: 206623

erxbout, 2025-01-17:
@tmcrash @gpitzer Ich kann eine neue Version direkt auf den charger einspielen heute bzw in den kommenden Tagen

Einfach mit 🚀  reagieren wenn das passt

erxbout, 2025-01-23:
@gpitzer bitte ;)

Über welche Schnittstelle kommt der fehler? Welches logfile?
Könnte sich um ein oversight handeln, da für gewöhnlich pre versions nicht ausgeliefert werden..

tmcrash, 2025-01-23:
@erxbout Meine Wallbox zeigt noch 56.8 an. Wann erfolgt das Upgrade?

erxbout, 2025-01-23:
@tmcrash Habe auf Rückmeldung gewartet, wollte das Update nicht einfach so aufspielen..
Ist jetzt aber drauf ;)

tmcrash, 2025-01-23:
@erxbout 
Ich hatte doch mit einer "Rakete" auf deinen Beitrag mit der Anfrage zum Update reagiert.
Macht aber nichts. Ist ja nun geklärt.
Vielen Dank für das Aufspielen.
Eben ist gerade ein Auto verbunden worden und das Laden wurde sofort von IOB gestoppt.
Dann habe ich adhoc Laden über IOB (VIS) aktiviert und es wurde gestartet. Läuft also alles wieder.
Die Register 204 und 206 stimmen auch (der Fix war schon nach 56.8 enthalten, aber wegen anderer Probleme nicht nutzbar).
Vielen Dank für die Hilfe.

erxbout, 2025-01-23:
Oh das hab ich dann falsch gesehen ich dachte das war [gpitzer](https://github.com/gpitzer) 
My bad ^^

Aber super, dass es nun funkioniert 👍

gpitzer, 2025-01-23:
@erxbout ich habe auch die 🚀 gezündet.
ich bekomme die Fehlermeldung im ioBroker

[image]

Und ich kann auch keine anderen Werte wie die aktuell geladene Leistung auslesen.

gpitzer, 2025-01-24:
@erxbout 
Anscheinend kann die ioBroker diese Werte dann nicht auslesen. Schreiben, was auch wichter für mich ist funktioniert ja.
Ich werde mal versuchen ob ich das über ein Ticket bei ioBroker lösen kann, ansonsten warte ich auf die nächste FW.
Vielen Dank auf jeden Fall!!!

0xFEEDC0DE64, 2025-02-03:
seit letztem Donnerstag gibts eine neue beta firmware version mit gefixtem modbus, sind damit alle probleme behoben?

gpitzer, 2025-02-04:
> seit letztem Donnerstag gibts eine neue beta firmware version mit gefixtem modbus, sind damit alle probleme behoben?

scheint zu funktionieren 👍

erxbout, 2025-02-05:
Fixed in 58.1

pholler, 2025-02-15:
Sorry to say but the issue is not fixed in 58.1 Beta. Just tried it and there is still no way to write registers!

CommanderRedYT, 2025-02-19:
@pholler Hi, wie genau hast du es getestet? Habe gerade mit einem Debugging-Tool (https://github.com/0xFEEDC0DE64/qtmodbustester) es getestet, und da geht alles. (Register 299 mit type "Holding Register")

Folgende API keys haben folgende Werte:
| Key | Value |
|:---:|:-----:|
|men|true|
|msb|false|
|msr|true|

jurassicparkicecream, 2025-04-16:
Hi @erxbout in 57.1 besteht das Modbus Problem weiterhin oder? Würde gerne updaten, da ich auch immer wieder WLAN Abbrüche habe, aber traue mich nicht ;)

Ansonsten nehme ich auch gerne die Beta, wenn dort Modbus läuft.  Seriennummer: 219740

jurassicparkicecream, 2025-05-20:
Danke @erxbout , Update auf 59.0 ist angekommen und Modbus schreiben funktioniert wieder!

erxbout, 2025-05-21:
58.1 Beta ist nicht mehr Verfügbar aufgrund von anderen Bugs

59.x Beta ist auf dem Weg aber da gibt es noch keinen Zeithorizont wann das ausgerollt wird

Cool, dass das update gut angekommen ist, hoffentlich löst das Probleme ;)

kenjibro, 2025-07-28:
Bei mir funktioniert das Schreiben der Modbus-Werte mit der 59.4 bisher problemlos.

closed 2025-07-28 by pd82wi

pd82wi, 2025-11-02:
Bei einem Phasenwechsel stoppt der Ladevorgang, dann macht der go-e die Phasenumschaltung und startet dann den Ladevorgang wieder. Ich steuere den go-e über Loxone und habe die Logik abhängig von der Überschussleistung der PV-Anlage programmiert: 
- PV-Leistung > 4 kW für die Dauer von 15min --> Umschaltung in 3-phasigen Betrieb
- PV-Leistung < 3,7kW für die Dauer von 15min --> Umschaltung in 1-phasigen Betrieb

Damit wird maximal alle 15min umgeschaltet. An sonnigen Tagen mit vielen durchziehenden Wolken, wodurch diese Logik aufgrund dynamischer Überschussleistung evtl. häufiger schalten würde, stelle ich eine fixe Ladeleistung ein und akzeptiere, dass dann PV-Überschuss teilweise ungenutzt, bzw. Strom aus dem Netz entnommen wird. An sonnigen Tagen mit nur wenigen Wolken funktioniert diese Logik sehr gut.

pd82wi, 2025-11-02:
Die Einstellung Automatisch habe ich bisher nicht verwendet.

Ich verwende Firmware 59.4, damit funktioniert die Phasenumschaltung bisher problemlos.

tmcrash, 2025-11-07:
Bei mir ist vor einiger Zeit eine spezielle Version von 57.1 installiert worden. Ich bekomme seitdem keine höhere Version angeboten. Wie kommt man an Version 59.4 ran?

erxbout, 2025-11-08:
59.4 wird nicht an v3 hardware ausgeliefert aufgrund von Bugs die explizit v3 betreffen.

In 60.0 behoben und die sollte ganz normal verfügbar sein (derzeit beta)