goecharger/go-eCharger-API-v2 #255: Netzbezug begrenzen (Leistung)

motze92, 2024-12-19:
Gibt es eine Möglichkeit die Leistung für den Netzbezug zu begrenzen?

Ich sende an pPGrid meine aktuelle Leistung vom Zähler. Möchte aber, dass z.b. maximal 5kW vom Netzt bezogen werden und sich die Ladestation danach regelt.

Es sollte die Leistung sein, die begrenzt und nicht die Stromstärke, da in Italien der Haushaltsstrom auf die Leistung begrenzt ist. (Üblicherweise 4,5kW + kurzzeitig 30% Toleranz, ob ich das dann über eine Phase beziehe oder aufgeteilt über 3 Phasen ist egal)

schinagl, 2024-12-21:
Du kannst dazu pgt verwenden

MiOrt, 2025-01-06:
Darf ich mich kurz einhängen? Ich denke nicht, dass das so funktioniert. Ich denke, dass pgt nur für die PV-Regelung als Regelpunkt herangezogen wird. Ich hab Zielwert Netz derzeit auf 0. Wenn ich Basic lade, bietet der Go-e aber dem EV den eingestellten Maximalstrom an. Somit dürfte das der falsche API-Key sein. Ich habe auch noch keinen Key gefunden, der das gewünschte bewirkt. Entweder pgt für den Regelpunkt bei Überschussladen, oder man begrenzt den Strom des Go-e egal in welchem Modus. Mit einem derartigen Key wäre ja auch sofort problemlos die Umsetzung des §14a möglich.

bruma59, 2025-02-24:
I've the same problem in Belgium, in the standard connection price for electricity is 625Wh/15min included, each 250Wh/15min above that value is 5€ surplus. Each month there is a reset. One kwarter of an hour peaks determines the (surplus use) cost of the month.

Until now I've used the pgt to limit the kWh rate and it works (if I put 300W less in the pgt than what I want as a limit due to overshoot ex pgt=3200Wh for a limit of 3500Wh/4 = 875Wh/15min)

For each charge session I've to decide to change the pgt if I want to charge with a grid use limit or with surplus PV use. When I forget it maybe the car is not charged (have to use a DC charger) or I've to pay for the grid current while there was enough surplus current, most likely in the morning or evening when PV is less than the pgt value.

So a paramter to limit the Wh used, like the pgt for the non-surplus charging conditions would be welcome. 

I see in the future that the cost of grid use will be time related, in Belgium there is a call for objections to use 3 timeframes for the winter and 2 timeframes for the rest. So that 5€/250Wh/15min will be different for the different timeslots probabely more expensive for peak periods

MiOrt, 2025-03-19:
Ich hatte mich diesbzgl. - gerade auch wegen §14a EnWG in Deutschland - an den Support gewandt, weil ich denke, dass viele Käufer nicht wissen, dass die beworbene und aktuell implementierte "netzdienlichen Leistungsregelung durch den Stromnetzbetreiber" im Grunde nichts mit intelligent oder smart zu tun hat. Unabhängig des PV-Überschuss, wird stumpf die Ladeleistung heruntergeregelt. Ich schrieb vor einigen Wochen das Nachfolgende (natürlich muss der ggf. notwendige Gleichzeitigkeitsfaktor noch berücksichtigt werden, dass dürfte aber über das EMS einfach und individuell zu implementieren sein), bis auf die "Danke, werden wir an die Entwickler geben"-Antwort gab es bislang kein Feedback:

> Die Verwendung des API-key ama ist die simpelste Form der Dimmung nach §14a EnWG. Ein einfache Reduzierung des maximalen Ladestroms auf 6A dürfte bei der Mehrzahl der auf dem Markt befindlichen Wallboxen die Herangehensweise an die gesetzliche Forderung sein. Den wenigsten Nutzern dürfte dabei bewusst sein, dass dadurch die Dimmung auch erfolgt, wenn im Grunde ausreichend PV-Überschuss vorhanden wird. Zudem findet die Anzahl der tatsächlich verwendeten Phasen keine Beachtung. EV, die nur mit einer Phase laden können, laden demnach im Fall der Dimmung nur noch mit rund 1,38kW, anstelle der erlaubten 4,2kW.
>
>Diese Form der Dimmung ist auch nicht im Sinne des Gesetzgebers. Um unnötige Angst vor dem Gesetz zu vermeiden, klärt die Bundesnetzagentur hierzu explizit auf: 
>
>„Es kommt ausschließlich auf den Leistungsbezug aus dem Stromnetz an. Selbst produzierter Strom wird nicht mit der Leistungsgrenze von 4,2 kW verrechnet.
>
>Es handelt sich um einen technologieoffenen Ansatz: Die Leistung mehrerer Anlagen im Haushalt kann durch Energiemanagementsysteme verrechnet werden. Eine Wallbox darf also zum Beispiel im Falle einer Netzbetreibersteuerung mehr Strom beziehen, wenn dieser aus der eigenen Solaranlage bezogen wird. Lediglich der zulässige Strombezug aus dem Verteilernetz darf nicht überschritten werden.“
>
>Zusätzlich wird ausgeführt:
>
>„Wie hoch die Mindestleistung ist, hängt davon ab, wie Ihre steuerbare Verbrauchseinrichtung angesteuert wird: Per Direktansteuerung oder mittels Energie-Management-System (EMS). Zwischen diesen beiden Möglichkeiten entscheiden Sie individuell, also pro steuerbarer Verbrauchseinrichtung hinter einem Netzanschluss.
>
>Direktansteuerung
>Die Mindestleistung beträgt pro steuerbarer Verbrauchseinrichtung grundsätzlich 4,2 kW.
>
>Steuerung mittels EMS
>Für alle steuerbaren Verbrauchseinrichtungen ist die Mindestleistung unter Berücksichtigung eines vorgegebenen Gleichzeitigkeitsfaktors vom Netzbetreiber gesamthaft zu ermitteln. Den ermittelten Sollwert für die Mindestleistung können Sie über das EMS auf Ihre Verbrauchseinrichtungen verteilen. Selbst produzierter Strom kann berücksichtigt werden.“
>
>Wobei die Regelung über den Gleichzeitigkeitsfaktor hier zunächst keine Rolle spielen soll, dies wäre ggf. ein Thema für den go-e Controller. Sollte mein Vorschlag umgesetzt werden, wäre selbst die Gleichzeitigkeitsregelung problemlos über den neuen key mdimp möglich.
>
>(https://www.bundesnetzagentur.de/DE/Vportal/Energie/SteuerbareVBE/start.html)
>
>Ich schlage folgende Vorgehensweise vor, um die Dimmung nach §14a EnWG für den Nutzer des go-e mit maximalem Nutzen umzusetzen:
>
>1) Einführung eines neuen API-key „mdimp“, R/W, int, Config, „maximum power from grid during dimming“ (vorgelegt mit 4200 für Deutschland)
>
>2) Einführung eines neuen API-key „dim“, R/W, bool, Config, „dimming power from grid on/off“
>
>3) Einbau der neuen keys in die Regelschleife des go-e. Wenn in einem Modus PV-Überschuss genutzt wird, muss in diesem Fall - wenn dim = true - anstelle pgt der key mdimp beachtet werden, wenn mdimp<pgt. In allen anderen Situationen muss der go-e unter Beachtung der tatsächlich genutzten Phasenzahl pnp den Strom entsprechend begrenzen (anstelle ama den Wert 6A bei pnp=3 und 18A bei pnp=1 (bei 11kW-Variante dann 16A)). Somit dürfte sichergestellt sein, dass der Netzbezug des go-e nie über dem Wert mdimp liegt.
>
>Ich hoffe, ich sehe die Sache nicht zu einfach. Im Grunde liegen alle Parameter bis auf die neuen keys bereits vor und eine Anpassung der Regelschleife dürfte keine große Herausforderung sein. Natürlich kann man das Ganze auch über Home Assistant oder andere Systeme individuell implementieren. Dies würde aus meiner Sicht jedoch dem Anspruch an eine smarten/intelligenten Wallbox, wie sie derzeit vermarktet wird, wesentlich besser gerecht werden. Außerdem denke ich, dass es einige negative Rückmeldungen geben könnte, wenn tatsächlich mal jemand gedimmt wird und dann trotz PV und Überschuss das EV anstelle mit 11kW nur noch mit 4,2kW lädt.